margelo/react-native-vision-camera #1455: 🐛 [iPadOS] If app is set to landscape only, vision camera gets preview orientation wrong

zzz08900, 2023-02-06:
### What were you trying to do?

We have an iPadOS-only app that only supports landscape orientation. We tried to use react-native-vision-camera@2.15.4 with this app but found the preview is displayed with wrong orientation - it was rotated 90 degrees.
So when the iPad moves left and right, the image in react-native-vision-camera preview moves up and down.

### What happened instead?

With an app that only supports landscape orientation, when the iPad moves left and right, the image in react-native-vision-camera preview moves up and down.

### Device

iPad (9th Generation)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2023-02-06:
This bug is not present in version 2.14.x, I highly suspect this piece of code is what caused this issue
[image]

Before 2.15.3 there were codes that detects current orientation on camera launch but after this commit, it seems that react-native-vision-camera assumes the device is portrait on camera launch, which caused our issue.

I'll fiddle around a little more and return with what happened later.

JackHull, 2023-02-06:
Can confirm this issue with newest version.

mrousavy, 2023-02-14:
Can you record a video (with another phone) and post that here?

zzz08900, 2023-02-17:
> Can you record a video (with another phone) and post that here?

Sorry I got my hands really busy these days - almost overwhelmed by customer demands. Will do this weekend.

zzz08900, 2023-02-20:
https://user-images.githubusercontent.com/51951532/220184458-6b9c9427-3ef9-4f82-ac97-4ca9fc7134c9.MOV

@mrousavy here's the video. You can see from the texture of the background that the content of <Camera /> view is moving perpendicular to the movement direction of the iPad.

zzz08900, 2023-02-20:
Just tried adding back `private var windowInterfaceOrientation` which was removed in commit `bf71901` and nothing changed.

Do I need a full `delete everything and rebuild` after modifying files from react-native-vision-camera to make changes take effect? I got a feeling that Apple-made stuff likes to store all kinds of strange cache everywhere.

UmerWarraich95, 2023-02-24:
Hey @zzz08900   can you look into this issue as well and guide me to solve it?
https://github.com/mrousavy/react-native-vision-camera/issues/1445

I am facing this issue on IOS camera rotation but on Android it is working fine.

UmerWarraich95, 2023-03-03:
@sekcan  I have resolved this IOS rotation issue by downgrading the react-native-vision-camera to "2.14.1"
After downgrading, my IOS build still works and I have successfully uploaded it to testflight.

sekhuat, 2023-03-17:
I also facing the same issue, will try to downgrade to 2.14.1 if no other solution :(

fdalmard, 2023-04-06:
I had the same problem when using version 2.15.4.
Downgrading to version 2.15.2 fixed the issue 😉

lucksp, 2023-04-22:
> I had the same problem when using version 2.15.4. Downgrading to version 2.15.2 fixed the issue 😉

I was on version `^2.15.4` and noticing the `orientation={'portrait'}` prop doesn't seem to work when on landscape mode, in terms of how the picture is captured.  
I downgraded to 2.15.2 & 2.15.1 but had no success with the orientation changing.  MAYBE i am doing something wrong?

```
<ReanimatedCamera
                  // Do not add enableZoomGesture since using the Animated lib
                  // enableZoomGesture
                  style={{ ...styles.cameraPreview }}
                  ref={cameraRef}
                  photo
                  device={device}
                  isActive={isActive}
                  format={formats?.[0]}
                  animatedProps={animatedProps}
                  orientation={'portrait'}
                />
```

Still results as landscape when viewing final image on web too
[image]

fdalmard, 2023-04-23:
> I was on version `^2.15.4` and noticing the `orientation={'portrait'}` prop doesn't seem to work when on landscape mode, in terms of how the picture is captured. I downgraded to 2.15.2 & 2.15.1 but [...]

The `orientation` prop didn't work for me too. Here what worked for me:
- I removed the `orientation` prop
- I used https://github.com/wonday/react-native-orientation-locker to force the screen orientation to either portrait (`lockToPortrait` API) or landscape (`lockToLandscapeLeft` API)

After that, the camera orientation was working as I wanted

```js
<Camera
  ref={cameraRef}
  style={{ flex: 1 }}
  device={device}
  isActive
  photo
  enableZoomGesture
/>
```

lucksp, 2023-04-24:
thank you for suggestion @fdalmard 
The thing is, we want to allow landscape mode (especially for video with wide angle), but we need the image to be saved in portrait orientation, so that viewing the result on device or on web is viewable in a portrait mode.

umbertoghio, 2023-04-25:
Hi, I have the same issue on iPhone (android working fine), I've downgraded to version 2.15.2 and the issue is solved!

Basically the rotation is 90deg wrong when landscape, it's fine when portrait, I'm unlocking screen orientation in order to let user chose portrait or landscape videos.

I tried to play with the "orientation" prop, from the swift code it seems that if present it should ignore device orientation but it made no difference, even when setting the prop dynamically on rotation event (from JS).

Hope it can help

lucksp, 2023-04-25:
i changed to `2.15.2` and the `orientation` is still stuck at `6` when taking pictures in both Vertical & Horizontal orientation

lucksp, 2023-04-25:
> * I removed the `orientation` prop
> * I wrapped the Camera with a Modal with `supportedOrientations={['portrait', 'landscape']}`
> * I used https://github.com/wonday/react-native-orientation-locker to force the screen orientation to either portrait (`lockToPortrait` API) or landscape (`lockToLandscapeLeft` API)
> 
> After that, the camera orientation was working as expected
> 
> ```js
> <Modal
>   onRequestClose={() => {}}
>   visible
>   supportedOrientations={['portrait', 'landscape']}
> >
>   <Camera
>     ref={cameraRef}
>     style={{ flex: 1 }}
>     device={device}
>     isActive
>     photo
>     enableZoomGesture
>   />
> </Modal>
> ```

Thank you @fdalmard for the suggestion.  With quite a bit of custom logic, this now works when capturing an image:

```
const capturePhoto = async () => {
    if (cameraRef?.current && !snapping) {
      setSnapping(true);
      const photo = await cameraRef.current.takePhoto({
        flash: flashMode,
      });

      let rotation = 0;

      if (orientation === OrientationType['LANDSCAPE-LEFT']) {
        rotation = -90;
      } else if (orientation === OrientationType['LANDSCAPE-RIGHT']) {
        rotation = 90;
      } else if (orientation === OrientationType['PORTRAIT-UPSIDEDOWN']) {
        rotation = 180;
      }

     // now use `expo-image-manipulator`
      const photoWithOrientation = await manipulateImage({
        imageUri: photo.path,
        actions: [{ rotate: rotation }],
        saveOptions: {
          compress: 1,
        },
      });

      handleSaveAsset([
        {
          path: photoWithOrientation.uri,
          width: photoWithOrientation.width,
          height: photoWithOrientation.height,
          isRawPhoto: false,
        },
      ]);
    }
  };
```

valn1, 2023-08-28:
did anyone find a solution to this?
the modal thing didn't really work for me and i don't think we should need to use that
the ios's camera should work the same as android's

valn1, 2023-08-28:
the prop `orientation` does nothing, i wrote a setInterval to cycle through all four possibilities and the preview just blinks for a moment and logs `VisionCamera.didSetProps(_:): Updating 1 prop(s)...`
note that my app has every orientation active in the deployment info config section but for this screen i use react-native-orientation-locker to `Orientation.lockToLandscape();`
right now i am still testing stuff and trying to fix it, will bring back more information as i go

valn1, 2023-08-29:
i could not fix it the way i wanted to, but made a quick workaround.

just add the patch to your project and add the prop `orientation={'portraitUpsideDown'}` to your camera element
[react-native-vision-camera+2.15.6.patch](https://github.com/mrousavy/react-native-vision-camera/files/12468775/react-native-vision-camera%2B2.15.6.patch)

ps. i know it is not pretty, but i have deadlines

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

FabianTrain, 2023-09-30:
> Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.
> 
> If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

I am recently upgraded to V3 and this issue came up. V2.15 was working for me.

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.

umbertoghio, 2023-10-01:
> This bug is not present in version 2.14.x, I highly suspect this piece of code is what causing this issue. [image]
> 
> Before 2.15.3 there were codes that detects current orientation on camera launch but after this commit, it seems that react-native-vision-camera assumes the device is portrait on camera launch, which caused our issue.
> 
> I'll fiddle around a little more and return with what happened later.

Thanks to your comment I managed to resolve the orientation issue on 2.x version by forking the repository and replace iOS/CameraView+Orientation.switft with 

```
//
//  CameraView+Orientation.swift
//  VisionCamera
//
//  Created by Marc Rousavy on 04.01.22.
//  Copyright © 2022 mrousavy. All rights reserved.
//

import Foundation
import UIKit

extension CameraView {
  /// Returns the current _interface_ orientation of the main window
  private var windowInterfaceOrientation: UIInterfaceOrientation {
    if #available(iOS 13.0, *) {
      return UIApplication.shared.windows.first?.windowScene?.interfaceOrientation ?? .unknown
    } else {
      return UIApplication.shared.statusBarOrientation
    }
  }

  /// Orientation of the input connection (preview)
  private var inputOrientation: UIInterfaceOrientation {
    return windowInterfaceOrientation
  }

  // Orientation of the output connections (photo, video, frame processor)
  private var outputOrientation: UIInterfaceOrientation {
    if let userOrientation = orientation as String?,
       let parsedOrientation = try? UIInterfaceOrientation(withString: userOrientation) {
      // user is overriding output orientation
      return parsedOrientation
    } else {
      // use same as input orientation
      return inputOrientation
    }
  }

  internal func updateOrientation() {
    // Updates the Orientation for all rotable connections (outputs) as well as for the preview layer
    DispatchQueue.main.async {
      // `windowInterfaceOrientation` and `videoPreviewLayer` should only be accessed from UI thread
      let isMirrored = self.videoDeviceInput?.device.position == .front

      self.videoPreviewLayer.connection?.setInterfaceOrientation(self.inputOrientation)

      let connectionOrientation = self.outputOrientation
      self.cameraQueue.async {
        // Run those updates on cameraQueue since they can be blocking.
        self.captureSession.outputs.forEach { output in
          output.connections.forEach { connection in
            if connection.isVideoMirroringSupported {
              connection.automaticallyAdjustsVideoMirroring = false
              connection.isVideoMirrored = isMirrored
            }
            connection.setInterfaceOrientation(connectionOrientation)
          }
        }
      }
    }
  }
}

```

andrespompiglio, 2023-10-19:
> Thanks to your comment I managed to resolve the orientation issue on 2.x version by forking the repository and replace iOS/CameraView+Orientation.switft with
>
> ```
> //
> //  CameraView+Orientation.swift [...]

Thanks!, we've applied above change to version 2.16.3 and it worked ok!

garmoshka-mo, 2023-11-29:
there is still issue on iPads - if you will try to rotate device while camera is open - it will turn image improperly.

You need to add this to `outputOrientation` to fix camera on iPads:

```
  private var outputOrientation: UIInterfaceOrientation {
    if (UIDevice.current.userInterfaceIdiom == .pad) {
        return inputOrientation
    }
```

PR #1544 (2023-09-30): fix: updated orientation calculations
## What

This PR should fix the issues in video orientation (preview and recording) being "fixed" to portrait orientation regardless of device orientation in iOS.

Adapted from https://stackoverflow.com/a/36575423

## Changes

* Changed the CameraView orientation extension so that it takes the UIDevice's reported orientation if the userOrientation is not provided
* On rotation change it will update the preview layer orientation and update the preview layer bounds accordingly

## Tested on

* iPhone 14 Pro
* iPad Mini

## Related issues

* Fixes #1539 
* Fi [... cut]