margelo/react-native-vision-camera #1459: Does react-native-vision-camera support brightness and exposure❓

nikita-usichenko, 2023-02-07:
### Question

I attempted to find a solution for changing the brightness and exposure in the react-native-vision-camera library, but I haven't been successful. Can anyone enlighten me on whether it is possible to make these adjustments and, if so, how I can go about it? Are there any alternative libraries that offer these options?

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Ha0ran2001, 2023-04-20:
I also wanna know how to set exposure、ISO、focal length、white balance in react-native-version-camera.

mrousavy, 2023-09-30:
Hey - not yet, but I think there's a feature request somewhere. Once enough people sponsor that, I can implement it

closed 2023-09-30 by mrousavy