margelo/react-native-vision-camera #1460: Build failed with an exception on RN 0.71.2🐛

Productivix, 2023-02-07:
### What were you trying to do?

use this lib : simple test

### Reproduceable Code

```tsx
simple, as usual :

npx react-native init testrn71
cd testrn71
yarn add react-native-vision-camera
yarn run android
```

### What happened instead?

build failed 

### Relevant log output

```shell
FAILURE: Build failed with an exception.

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30
```

### Device

android 8.1

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chernandezlt, 2023-02-08:
In your android/build.gradle add kotlinVersion variable

```
buildscript {
    ext {
        ...
        kotlinVersion = "1.6.20"
    }
}
```

Productivix, 2023-02-08:
thank you : simple , it works  !

Productivix, 2023-06-29:
just to help : now in RN 0.72, put  : kotlinVersion = "1.7.22" , so you have 1 virtual Java machine less at complilation !