margelo/react-native-vision-camera #1462: Camera.getAvailableCameraDevices() returns undefined on Android, physical device🐛

stivenm16, 2023-02-08:
### What were you trying to do?

I'm trying to access to all available Camera Devices but it returns null when I'm running my app on a physical device, however when  I run the code in the emulator it works fine and returns a full array with options. The phone is connected via usb. This unexpected behaviour still persist after build it

### Reproduceable Code

```tsx
const TestComponent = () => {
  const { navigate } = useNavigation();
 
  useEffect(() => {
    const requestPermissions = async () => {
      await Camera.requestCameraPermission();
    };
    requestPermissions();
  });

  const devices = useCameraDevices();
  const device = devices.back;

  // console.log(devices, "devices"); => it returns null on physical device

  if (device == null)
    return (
      <View>
        <Text> Device null</Text>
      </View>
    );

  return (
    <View style={{ flex: 1, backgroundColor: "red", padding: 10 }}>
      <Camera style={{ flex: 1 }} device={device} isActive={true} />
      <Text>It works</Text>
    </View>
  );
};
```

### What happened instead?

useCameraDevices() returns null when is called on a physical device

### Device

Silver Max lite 2

### VisionCamera Version

2.15.2 - 2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EmiBenitez, 2023-03-02:
any news about this? I have the same problem.

Ntouris3, 2023-03-17:
I'm facing the same issue

chj-damon, 2023-03-20:
I tried in `Smartisan` which is another brand and also the devices are all undefined.

closed 2023-09-01 by mrousavy