margelo/react-native-vision-camera #1463: 🐛 No implementation found for void CameraView.frameProcessorCallback

chernandezlt, 2023-02-08:
### What were you trying to do?

I'm upgrading our react native version 0.67.5 to 0.71.1, the mobile app runs well, but I have a problem with the frame processor when I try to read a barcode.

### What happened instead?

The app display an error (frame processor aren't enabled).

[image]

When I'm building the app the console displays that the frame processor are enabled.

[image]

### Relevant log output

```shell
I've been checking the logs and I've found this issue but I don't what is causing it.

com.envasetechnologies.envasemobile  E  FATAL EXCEPTION: pool-50-thread-1
                                                                                                    Process: com.envasetechnologies.envasemobile, PID: 22634
                                                                                                    java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
                                                                                                    	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
                                                                                                    	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:491)
                                                                                                    	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:463)
                                                                                                    	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$androidx-camera-core-ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
                                                                                                    	at java.lang.Thread.run(Thread.java:923)
2023-02-07 18:39:52.430 22634-24695 AndroidRuntime          com.envasetechnologies.envasemobile  E  FATAL EXCEPTION: pool-50-thread-2
                                                                                                    Process: com.envasetechnologies.envasemobile, PID: 22634
                                                                                                    java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
                                                                                                    	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
                                                                                                    	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:491)
                                                                                                    	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:463)
                                                                                                    	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$androidx-camera-core-ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
                                                                                                    	at java.lang.Thread.run(Thread.java:923)
2023-02-07 18:40:00.561 22634-24714 AndroidRuntime          com.envasetechnologies.envasemobile  E  FATAL EXCEPTION: pool-50-thread-3
                                                                                                    Process: com.envasetechnologies.envasemobile, PID: 22634
                                                                                                    java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
                                                                                                    	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
                                                                                                    	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:491)
                                                                                                    	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:463)
                                                                                                    	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda2.analyze(Unknown Source:2)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$androidx-camera-core-ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283)
                                                                                                    	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
[... cut ...]
Xiaomi Mi9, Samsung S22

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

marouane-edghoughi, 2023-02-09:
Hello, did you manage to solve this issue?

chernandezlt, 2023-02-10:
> Hello, did you manage to solve this issue?

No yet, the only solution was to disable the frame processors, but we need to scan barcodes.

chiika, 2023-02-11:
I just had the same issue yesterday while upgrading RN 0.70.6 to 0.71.2 and stumbled over this issue report.
I used to have Hermes disabled because it caused issues back then, so I also disabled it while upgrading. After reenabling Hermes, this frame processor runtime link issues went away. I hope it helps.

PopBot, 2023-03-28:
@chiika Hm, I have enabled Hermes, and it still doesn't work for me. Would you mind sharing your `package.json`? Thanks.

chiika, 2023-03-28:
> @chiika Hm, I have enabled Hermes, and it still doesn't work for me. Would you mind sharing your `package.json`? Thanks.

Not at all, here you go:
(one comment tho, I'm only doing android builds right now)
```
{
  "name": "scanner",
  "version": "1.0.5",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "@stripe/stripe-terminal-react-native": "^0.0.1-beta.11",
    "@react-native-async-storage/async-storage": "^1.17.12",
    "@react-native-community/netinfo": "^9.3.7",
    "@react-navigation/material-bottom-tabs": "^6.2.15",
    "@react-navigation/material-top-tabs": "^6.6.2",
    "@react-navigation/native": "^6.1.6",
    "@react-navigation/stack": "^6.3.16",
    "axios": "^1.3.4",
    "color": "^4.2.3",
    "moment": "2.29.4",
    "react": "18.2.0",
    "react-native": "0.71.4",
    "react-native-config": "^1.5.0",
    "react-native-device-info": "^10.4.0",
    "react-native-gesture-handler": "2.9.0",
    "react-native-nfc-manager": "^3.14.2",
    "react-native-pager-view": "6.2.0",
    "react-native-paper": "^5.5.1",
    "react-native-reanimated": "2.14.4",
    "react-native-safe-area-context": "^4.5.0",
    "react-native-screens": "^3.20.0",
    "react-native-sqlite-storage": "^6.0.1",
    "react-native-svg": "^13.8.0",
    "react-native-vector-icons": "^9.2.0",
    "react-native-vision-camera": "^2.15.4",
    "vision-camera-code-scanner": "0.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.21.3",
    "@babel/preset-env": "7.20.2",
    "@babel/runtime": "^7.21.0",
    "@react-native-community/eslint-config": "^3.2.0",
    "@tsconfig/react-native": "^2.0.3",
    "@types/color": "^3.0.3",
    "@types/jest": "^29.5.0",
    "@types/react": "^18.0.29",
    "@types/react-native": "^0.71.5",
    "@types/react-native-sqlite-storage": "^6.0.0",
    "@types/react-native-vector-icons": "^6.4.13",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.5.0",
    "babel-plugin-inline-import": "^3.0.0",
    "eslint": "^8.36.0",
    "jest": "^29.5.0",
    "metro-react-native-babel-preset": "0.76.0",
    "prettier": "^2.8.7",
    "react-test-renderer": "18.2.0",
    "stripe": "^11.16.0",
    "typescript": "^5.0.2"
  },
  "jest": {
    "preset": "react-native",
    "moduleFileExtensions": [
      "ts",
      "tsx",
      "js",
      "jsx",
      "json",
      "node"
    ]
  }
}
```

using react-native 0.71.4 right now, build.gradle is mostly unchanged except I've added a 'staging' flavor to build types.
here is gradle.properties:
```
# Project-wide Gradle settings.

# IDE (e.g. Android Studio) users:
# Gradle settings configured through the IDE *will override*
# any settings specified in this file.

# For more details on how to configure your build environment visit
# http://www.gradle.org/docs/current/userguide/build_environment.html

# Specifies the JVM arguments used for the daemon process.
# The setting is particularly useful for tweaking memory settings.
# Default value: -Xmx512m -XX:MaxMetaspaceSize=256m
org.gradle.jvmargs=-Xmx2048m -XX:MaxMetaspaceSize=512m

# When configured, Gradle will run in incubating parallel mode.
# This option should only be used with decoupled projects. More details, visit
# http://www.gradle.org/docs/current/userguide/multi_project_builds.html#sec:decoupled_projects
# org.gradle.parallel=true

# AndroidX package structure to make it clearer which packages are bundled with the
# Android operating system, and which are packaged with your app's APK
# https://developer.android.com/topic/libraries/support-library/androidx-rn
android.useAndroidX=true
# Automatically convert third-party libraries to use AndroidX
android.enableJetifier=true

# Version of flipper SDK to use with React Native
FLIPPER_VERSION=0.125.0

# Use this property to specify which architecture you want to build.
# You can also override it from the CLI using
# ./gradlew <task> -PreactNativeArchitectures=x86_64
reactNativeArchitectures=armeabi-v7a,arm64-v8a,x86,x86_64

# Use this property to enable support to the new architecture.
# This will allow you to use TurboModules and the Fabric render in
# your application. You should enable this flag either if you want
# to write custom TurboModules/Fabric components OR use libraries that
# are providing them.
newArchEnabled=false

# Use this property to enable or disable the Hermes JS engine.
# If set to false, you will be using JSC instead.
hermesEnabled=true

```

PopBot, 2023-03-29:
Thanks so much @chiika ! I realized that I had enabled **Hermes** in my **Podfile** and not in the **gradle.properties**. After enabling it, I got the package working 😄

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy