margelo/react-native-vision-camera #1464: https://mrousavy.com cannot be reached 🐛

nikitph, 2023-02-09:
### What were you trying to do?

I am trying to read the docs on https://mrousavy.com/react-native-vision-camera/docs/guides/ but the site is down

### What happened instead?

[image]

### Device

desktop browser

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-02-09:
It works for me, even after clearing cache

[image]

closed 2023-02-09 by mrousavy

nikitph, 2023-02-09:
It works for me now with cloudflare warp. I m not sure of the root cause but m glad its not a widespread problem