margelo/react-native-vision-camera #1469: frame-processor/unavailable: Frame Processors are not enabled🐛

marouane-edghoughi, 2023-02-09:
### What were you trying to do?

I built a face recognition app, the recognition is basically done in the backend. And I recently tried to also add this feature on the app also, so I used Tensorflow for that.

I tried resetting the cache and cleaning gradle, disabling debugger & JS Debug Mode, none of em works

### Reproduceable Code

```tsx
// Other code....

const [model, setModel] = useState(null)

    const loadFaceModel = async () => {
        // Load the Tensorflow.js face detection model
        const faceModel = await tfjs_converter.loadGraphModel(
            'https://tfhub.dev/tensorflow/tfjs-model/ssd_mobilenet_v1/1/default/1',
            {
                fromTFHub: true,
                onProgress: (fraction) => {
                    console.log('Loading model ' + (fraction * 100).toFixed(1) + '%')
                }
            }
        )

        setModel(faceModel)
    }

    const frameProcessor = useFrameProcessor((frame) => {
        'worklet'
        if (model) {
            (async () => {
                // Convert the camera frame to a Tensor
                const tensor = tfjs_core.FromPixels(frame)

                // Use the model to perform face detection
                const detectionResults = await model.executeAsync(tensor)

                // Extract the bounding boxes of the faces
                const boxes = detectionResults[0].dataSync()
                const scores = detectionResults[1].dataSync()
                const count = detectionResults[2].dataSync()[0]

                console.log('Number of faces detected: ' + count)
            })()
        }
    })

useEffect(() => {
        checkCameraPermission()
        loadFaceModel()
    }, [])

return (
{device == null && !model ? (
                    <Text>Loading</Text>
                ) : (
                            <Camera
                                ref={camera}
                                device={device}
                                isActive={isFocused}
                                frameProcessorsEnabled={true}
                                frameProcessor={frameProcessor}
                                frameProcessorFps={5}
                                photo={true}
                                enableZoomGesture={isBackCam ? true : false}
                            />
)
```

### What happened instead?

App crashes after opening the camera

### Relevant log output

```shell
ERROR  frame-processor/unavailable: Frame Processors are not enabled. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting, js engine: hermes
```

### Device

Android

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

marouane-edghoughi, 2023-02-13:
I finally got it working after upgrading React to 0.71.2 which is the latest version. I couldn't understand why I've been encountering this issue on React 0.70. So that was it ^_^

closed 2023-02-13 by marouane-edghoughi

mayurm-virtueinfo, 2023-02-28:
I have upgraded to 0.71.2 but still error exist in iOS i am stucked in this error for 2 weeks and did not get the solution yet.

ningrunting, 2023-03-03:
I use react-native 0.71.0 and vision camera 2.15.4 I has the issue on android

jonghelixpay, 2023-03-15:
Also encountering the issue on android

"react-native": "0.71.4"
"react-native-reanimated": "^2.10.0"
"react-native-vision-camera": "^2.15.4"
"vision-camera-code-scanner": "^0.2.0"

barghi, 2023-09-07:
same issue here:
```
"react-native": "0.72.0",
"react-native-vision-camera": "3.0.0-rc.10",
"react-native-worklets-core": "^0.2.0",
"react-native-reanimated": "^3.4.2",
```