margelo/react-native-vision-camera #1473: ✨ Enable photo HDR capability

Balaji-Rahunathan, 2023-02-14:
### What feature or enhancement are you suggesting?

Currently all formats shows only video HDR. It will be great if v3 has photo HDR capabilty

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

I need this feature. Currently there are no alternatives or workarounds for this.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-02-22:
On iOS, photo HDR is absolutely non-trivial - looks like you need to capture multiple and fuse the images together yourself, basically re-invent the HDR algorithm. No idea why they don't provide that. See [Is HDR capture a thing in iOS?](https://www.reddit.com/r/iOSProgramming/comments/778ypz/is_hdr_capture_a_thing_in_ios/)

mrousavy, 2023-02-22:
[hhmmmmm](https://developer.apple.com/documentation/avfoundation/photo_capture/capturing_still_and_live_photos/capturing_a_bracketed_photo_sequence)