margelo/react-native-vision-camera #1474: EPERM: operation not permitted, lstat, errno: -4048

WonderHeaven, 2023-02-14:
### What were you trying to do?

I am now trying to create simple Camera app by using react-native 0.71 + react-native-vision-camera on Windows OS .

### What happened instead?

I am now having troubles with Error: EPERM: operation not permitted, lstat error when I execute "npm start" command.
Here is detail error 

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
node:events:491
      throw er; // Unhandled 'error' event
      ^

Error: EPERM: operation not permitted, lstat 'D:\Dev\react\prjTest\node_modules\react-native-vision-camera\android\build\intermediates\cxx\Debug\3o3p5j5h\obj\arm64-v8a\libVisionCamera.so.tmpbe4b372'
Emitted 'error' event on NodeWatcher instance at:
    at D:\Dev\react\prjTest\node_modules\metro-file-map\src\watchers\NodeWatcher.js:235:14
    at callback (D:\Dev\react\prjTest\node_modules\graceful-fs\polyfills.js:306:20)
    at FSReqCallback.oncomplete (node:fs:207:21) {
  errno: -4048,
  code: 'EPERM',
  syscall: 'lstat',
  path: 'D:\\Dev\\react\\prjTest\\node_modules\\react-native-vision-camera\\android\\build\\intermediates\\cxx\\Debug\\3o3p5j5h\\obj\\arm64-v8a\\libVisionCamera.so.tmpbe4b372'
}

Node.js v18.12.1

Could anyone help me, please?
I am beginner for Mobile Application Development. I am quite familiar with it. Please advise.

Thank you very much in advanced.

### Device

Android 

### VisionCamera Version

latest version

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DanielGannage, 2023-02-14:
I keep getting this error on 0.71. However it changes depending on what I patch it to.

**Here's what I've tried:**
a) adding `buildscript { ext { kotlinVersion:"1.8.0"}}` to `android/build.gradle` OR change the kotlin version in the module's `gradle.properties`
b) adding `buildscript { ext { kotlinVersion:"1.7.0"}}` to `android/build.gradle` OR change the kotlin version in the module's `gradle.properties`
c) adding `buildscript { ext { kotlinVersion:"1.7.21"}}"` to `android/build.gradle` OR change the kotlin version in the module's `gradle.properties` as it corresponds to what I have in android studio.
d) adding/removing `externalNativeBuild{ cmake { version "3.22.1" } }` to `android/app/build.gradle` and downloading corresponding cmake version in android studio sdk manager
e) different combinations thereof

**I've done a node modules wipe and a `gradlew clean` when appropriate. Usually doing this will change the error.**
**Sometimes the error is a:** 
i) `Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.`
or
ii) `An old version of CMake is being used that cannot automatically detect compiler attributes.  Compiler identification is being bypassed.  Some values may be wrong or missing.  Update to CMake 3.19 or newer to use CMake's built-in compiler identification.`

**If I remove everything I've added it goes back to this original error:**
`The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30`

Scratching my head here. Let me know if you solve it!

zzz08900, 2023-02-14:
I don't think vision camera supports windows. You need another library for this sort of stuff.

zzz08900, 2023-02-14:
Also this could be related to other libraries you are using. Would you mind posting your package.json here?

On a side note, there were reports that vision camera does not get along well with firebase.

Bottom line, can you build/run vision camera with base RN template (you can make one by using react-native init, then install vision camera as the only additional dependency)?

DanielGannage, 2023-02-14:
@zzz08900 I've just tried adding `buildscript { ext { kotlinVersion:"1.8.0"}}` to `android/build.gradle` file. Then, I did a node `modules` wipe, `yarn` to reinstall node modules, then `gradlew clean` on the android folder. Then, after running the project on android it prints and loops on this one warning 100 times before failing:
` CMake Warning at C:/Users/DanielGannage/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/DanielGannage/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/6b4v6r4w/armeabi-v7a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
    CMakeLists.txt:1 (project)

  CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  ninja: error: manifest 'build.ninja' still dirty after 100 tries`

**Here is my package.json:**
`{
  "name": "mobile",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "@microsoft/signalr": "5.0.0",
    "@react-native-async-storage/async-storage": "^1.17.11",
    "@react-native-community/geolocation": "^3.0.5",
    "@react-native-community/masked-view": "^0.1.11",
    "@react-native-community/push-notification-ios": "^1.10.1",
    "@react-navigation/drawer": "^6.5.7",
    "@react-navigation/native": "^6.1.2",
    "@react-navigation/stack": "^6.3.11",
    "@types/": "react-navigation/native",
    "@types/lodash": "^4.14.191",
    "@types/react-native-push-notification": "^8.1.1",
    "@types/redux": "^3.6.0",
    "@types/signalr": "^2.4.0",
    "@voximplant/react-native-foreground-service": "^3.0.2",
    "appcenter": "^4.4.5",
    "events": "^3.3.0",
    "i": "^0.3.7",
    "jail-monkey": "^2.8.0",
    "lodash": "^4.17.21",
    "moment": "^2.29.4",
    "npm": "^9.3.0",
    "react": "^18.2.0",
    "react-moment": "^1.1.3",
    "react-native": "^0.71.1",
    "react-native-app-auth": "^6.4.3",
    "react-native-background-actions": "^3.0.0",
    "react-native-collapsible": "^1.6.0",
    "react-native-config": "1.4.11",
    "react-native-dropdown-picker": "^5.4.4",
    "react-native-element-dropdown": "^2.5.3",
    "react-native-elements": "^3.4.3",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-image-picker": "^5.0.0",
    "react-native-image-resizer": "^1.4.5",
    "react-native-inappbrowser-reborn": "^3.7.0",
    "react-native-keep-awake": "^4.0.0",
    "react-native-keyevent": "^0.3.0",
    "react-native-manage-call-logs": "^1.0.3",
    "react-native-map-link": "^2.10.2",
    "react-native-push-notification": "^8.1.1",
    "react-native-reanimated": "2.14.1",
    "react-native-safe-area-context": "^4.4.1",
    "react-native-screens": "^3.18.2",
    "react-native-signature-capture": "^0.4.12",
    "react-native-simple-toast": "^2.0.0",
    "react-native-sound": "^0.11.2",
    "react-native-torch": "^1.2.0",
    "react-native-tts": "^4.1.0",
    "react-native-vector-icons": "^9.2.0",
    "react-native-version-number": "^0.3.6",
    "react-native-vision-camera": "^2.15.4",
    "react-redux": "^8.0.5",
    "redux": "^4.2.0",
    "redux-saga": "^1.2.2",
    "rn-secure-storage": "^2.0.8",
    "vision-camera-code-scanner": "^0.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/preset-env": "^7.14.0",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^3.0.0",
    "@tsconfig/react-native": "^2.0.2",
    "@types/jest": "^29.2.1",
    "@types/react": "^18.0.24",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.2.1",
    "eslint": "^8.19.0",
    "jest": "^29.2.1",
    "metro-react-native-babel-preset": "0.73.5",
    "prettier": "^2.4.1",
    "react-test-renderer": "18.2.0",
    "typescript": "4.8.4"
  },
  "jest": {
    "preset": "react-native"
  }
}
`

zzz08900, 2023-02-14:
I would suggest you try vision camera and vision camera code scanner with fresh RN base template to make sure your build system is set up alright.

DanielGannage, 2023-02-14:
@zzz08900 I created a new react native project. You were right, bottom line, the project does run with only the vision camera module. I started adding modules 1 by 1 to see what breaks. 

`"react-native-reanimated": "^2.14.4"` is where it breaks. I only need it for runOnJs which is needed for the frameprocessor in code scanner within vision camera. Any idea which versions would be compatible or if there is a workaround?

zzz08900, 2023-02-14:
Strange, I have a few Apps deployed with vision camera 2.14.x and reanimated 2.14.4 they all builds fine - on Linux servers that is.

 I don't have windows build system set up so cannot check but I guess some's wrong with windows.

Maybe try an older version of reanimated, say below 2.10?

WonderHeaven, 2023-02-15:
@DanielGannage No problem at all, it was very nice to learn from all of your posts and replies and @zzz08900 too.
Thank you very much for great supportive.
By the way I still received the error.
[image]

I just noticed there is other warn messages at the prior step before above error. I am not sure whether there is/are any significant or not. 
[image]

Here is some parts of my codes.
#1. File Name: prjTest\App.tsx
import { NavigationContainer } from '@react-navigation/native';
import React, { useEffect, useState } from 'react';
import { createNativeStackNavigator } from '@react-navigation/native-stack';
import { PermissionsPage } from './PermissionsPage';
import { MediaPage } from './MediaPage';
import { CameraPage } from './CameraPage';
import type { Routes } from './Routes';
import { Camera, CameraPermissionStatus } from 'react-native-vision-camera';

#2. File Name:  prjText\android\build.gradle
buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 21
        compileSdkVersion = 33
        targetSdkVersion = 33

        // We use NDK 23 which has both M1 support and is the side-by-side NDK version from AGP.
        ndkVersion = "25.2.9519653"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:7.3.1")
        classpath("com.facebook.react:react-native-gradle-plugin")
    }
}

#3. File Name: prjTest\android\src\main\AndroidManifest.xml
<manifest xmlns:android="http://schemas.android.com/apk/res/android">

    <uses-permission android:name="android.permission.CAMERA" />

    
    <uses-permission android:name="android.permission.RECORD_AUDIO" />

    <uses-permission android:name="android.permission.INTERNET" />
   .....
</manifest>

#4. File Name: prjTest\node_modules\react-native-vision-camera\android\gradle.properties
VisionCamera_buildToolsVersion=30.0.0
VisionCamera_compileSdkVersion=31
VisionCamera_kotlinVersion=1.7.20
VisionCamera_targetSdkVersion=31
VisionCamera_ndkVersion=21.4.7075529
android.enableJetifier=true
android.useAndroidX=true

Please advice, what I do wrong

zzz08900, 2023-02-15:
@DanielGannage is there a specific reason for you to modify gradle.properties of vision camera? For better practice you should always try setting kotlin versions in your project's build.gradle first 

Also vision camera came with kotlin 1.5.x, some versions in the  1.6.x range happens to work but most 1.7.x simply causes more trouble. By the look of your package.json I didn't find anything that requires 1.7.x. If any of your dependency do need 1.7.x then you probably need to downgrade that.

nithronium, 2023-03-05:
I can indeed verify that this is an issue related to the file permissions (either the library is trying to create/move/delete/rename some file) which is not permitted so the compilation gives an error. I don't have a solution for this either while working on Windows PC. Even running the terminal in administrator mode didn't solve the issue.

nithronium, 2023-03-05:
Issue is probably caused by those new file lines on Windows;

https://github.com/mrousavy/react-native-vision-camera/blob/566eda239523fcbc6862fe94587745ebf519cf92/android/build.gradle#L239

svhein, 2023-03-07:
Have you tried `react-native start --reset-cache` instead of `npm start`?

nithronium, 2023-03-08:
> Have you tried `react-native start --reset-cache` instead of `npm start`?

The issue is related to the file permission on Windows I believe. I don't know about the OP but I've tried the following things and got no result - switched back to another camera provider.

* Change file permissions (including node_modules folder and project folder) and allow for read-write on all files
* Run the commands from the administrator CMD & administrator powershell
* Try to build first and then deploy
* Move files and folders to another directory with leveraged access

Sadly none of them worked. It's related to the file permissions and probably lstat command somewhere.

warborn, 2023-03-10:
I had the same error in my project on my windows PC, the terminal where I run `yarn start --reset-cache` crashes with the error, I noticed that if I run again the command, metro actually connects to the app and I can keep working but today I tried to see if I can get rid of the error.

I followed some of the advices here like downgrading my `kotlinVersion` from `1.7.0` to `1.6.20 (the minimum required)`, running my commands from a CMD with administrator privileges but none of them worked. Then I found this on StackOverflow (https://stackoverflow.com/a/72596995)  and seems to work, when start metro the error is not thrown anymore. I hope this can be helpful for others.

nithronium, 2023-03-10:
> Then I found this on StackOverflow (https://stackoverflow.com/a/72596995) and seems to work, when start metro the error is not thrown anymore. I hope this can be helpful for others.

This mentions removing the hermes which might not be the best option to do so. Instead of this using another camera provider can be more logical.

akashi979, 2023-03-20:
hi all, i have been facing the same issue ,can anyone suggest something, even i am considering to move on with another cam provider , i am new to RN infact mob dev itself, @nithronium  could you tell which other we can use, i read somewhere react-camera is not supported anymore , still can we use it ... thanks!

ronuelbmoralita, 2023-04-20:
Any updates to these bugs?

Sidneycarlinijr, 2023-05-08:
> [nkqdev](/nkqdev)

deleting node_modules -> npm i + npx react-native start --reset-cache made it. Thanks

thenxkk, 2023-06-01:
After try all of the method above mentioned, I still fail

thenxkk, 2023-06-02:
> This mentions removing the hermes which might not be the best option to do so. Instead of this using another camera provider can be more logical.

I am facing same issue , could I know what   another camera provider you use

ronuelbmoralita, 2023-06-02:
> I am facing same issue , could I know what another camera provider you use

@thenxkk 
Bro, I can do it. At first, I installed a library, and after that, I ran these commands

1. cd android && gradlew clean
2. yarn cache clean
3. yarn start ------------- This is where I get the error. I'm so used to debugging wireless that I forget to rebuild my projects.
4. npx react-native run-android --------- This command is the best practice to completely rebuild and re-render our projects.

It worked for me. Hopefully, it will work for you. bless you.

thenxkk, 2023-06-02:
> 

oh!!! It works , thanks bro, many thanks. "yarn cache clean"make it work . After "yarn cache clean" , I do nothing(don't press any) and wait for few second  , it works.
But  it still fail at the second time that without "yarn cache clean".

valery-lavrik, 2023-06-08:
I've been trying to solve this problem all day. Nothing helps. even clearing the cache.

```
"dependencies": {
    "@dhairyasharma/react-native-encryption": "^1.0.0",
    "@react-native-community/datetimepicker": "^7.0.1",
    "@react-native-community/geolocation": "github:valery-lavrik/react-native-geolocation",
    "@react-native-community/netinfo": "^9.3.9",
    "@react-native-community/progress-bar-android": "^1.0.5",
    "@react-navigation/drawer": "^6.6.2",
    "@react-navigation/native": "^6.1.6",
    "@react-navigation/native-stack": "^6.9.12",
    "axios": "^1.4.0",
    "moment": "^2.29.4",
    "react": "18.2.0",
    "react-native": "0.71.10",
    "react-native-blob-util": "^0.17.3",
    "react-native-encrypted-storage": "^4.0.3",
    "react-native-fingerprint-scanner": "github:valery-lavrik/react-native-fingerprint-scanner",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-reanimated": "2.14.4",
    "react-native-safe-area-context": "^4.5.2",
    "react-native-screens": "^3.20.0",
    "react-native-sqlite-storage": "^6.0.1",
    "react-native-video": "^6.0.0-alpha.6",
    "react-native-vision-camera": "^2.15.4",
    "react-native-webview": "~12.1.0",
    "vision-camera-code-scanner": "^0.2.0"
  },
```

Any other ideas?

shubham-bhatt, 2023-07-01:
I was facing same problem, resolved when used different port -->

For windows

correct way is to go to root directory and follow instructions

run cmd as administrator 
run `npm start -- --port=8888`

then open another md as administrator in root dir
run `npm run android -- --port=8888`

lekbit, 2023-07-19:
Not sure why, but it seems to have helped solve my problem.
1)add string to android/build.gradle 
`kotlinVersion = "1.8.10"`
2)delete folder node_modules
3)npm i
4)npx react-native run-android
Also run VS code (if you're working in it) as an administrator.

Silvman, 2023-08-08:
`npm start -- --reset-cache`

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy