margelo/react-native-vision-camera #1476: First time video recording always has a stutter 🐛

billychasen, 2023-02-16:
### What were you trying to do?

When I start recording a video, the first few frames have a very low frame rate and create a stutter effect. This only happens on the very first recording.

### Reproduceable Code

```tsx
I tried `useCameraDevices("wide-angle-camera")` to pick just one camera, but it still happens.
```

### What happened instead?

First 3 seconds of video have a very low frame rate.

### Device

iPhone 13 Pro (16.3.1)

### VisionCamera Version

2.15.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-02-22:
Is that in Release as well?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy