margelo/react-native-vision-camera #1477: The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.

gokhancivili, 2023-02-16:
### What were you trying to do?

I'm trying to make a barcode reader

### Reproduceable Code

```tsx
Use a blank react native project
```

### What happened instead?

Build Process crashes.

### Relevant log output

```shell
> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: C:\ReactNative\test\node_modules
VisionCamera: Building with Hermes...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are disabled because REA v2 does not exist. C++ part will not be built.
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30
```

### Device

Android Simulator

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ozgurbayram, 2023-02-16:
I have same issue, it happens when i upgrade rn version to 0.71.3.
I solved this issue in this pr [](https://github.com/mrousavy/react-native-vision-camera/pull/1478)

zzz08900, 2023-02-17:
> You should define kotlin version in your android/build.gradle file to `kotlinVersion = "1.6.20" `

That does solve the problem but I guess there should be some regression test and nobody is feeling like doing it(myself included)
I haven't heard/experienced any issue using 1.6.20 + vision camera for a while but I would rather believe I'm just being lucky.

khadeshyam, 2023-02-26:
To solve this issue, you can try upgrading the Kotlin Gradle plugin version in the react-native-vision-camera project to version 1.6.20 or higher, which is supported by the Android Gradle plugin version you are using.

Go to **node_modules/react-native-vision-camera/android/build.gradle**

inside dependencies

```
dependencies {
           ....
           - classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
           + classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.20"
           ....
  } 

```
remove classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"

add classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.20"

TomWq, 2023-03-14:
When is it going to be fixed? You have to change it manually every time

ospfranco, 2023-03-15:
You don't need to manually patch the version in node_modules, you can add the following attribute in the root build.gradle:

[image]

mj-jadhav, 2023-04-02:
@gokhancivili `react-native-vision-camera` has defined the dependency of `org.jetbrains.kotlin:kotlin-gradle-plugin` to version `1.5.30` but you can override this by adding the following in your `android/gradle.properties`:
```
kotlinVersion=1.6.20
```
I faced the same issue but the above change fixed it for me.

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

KeunwooPark, 2023-05-12:
I'm also facing the same problem, but I do not sure it is fine to modify files in native projects. They will be generated from `app.json` So any direct modifications might cause problems. Are there any other solutions for this issue? I tried modifying [build properties](https://docs.expo.dev/versions/latest/sdk/build-properties/) for `android.kotlinVersion`, but Expo did not correctly generate native project files and having the same error.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy