margelo/react-native-vision-camera #1480: The Mi 13 can't call the camera❓

OPEN-9, 2023-02-17:
### Question

[image]

### VisionCamera Version

"react-native-vision-camera": "2.13.0"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

OPEN-9, 2023-02-17:
The Mi 11 is callable

OPEN-9, 2023-02-17:
xiaomi13 doesn't work

zzz08900, 2023-02-17:
I bet MIUI 13 is the source of evil here. You should ask them to fix their custom system image.

mrousavy, 2023-02-22:
Sorry not enough details

closed 2023-02-22 by mrousavy