margelo/react-native-vision-camera #1483: ❓Pause >> Flip Camera >> StartRecording on S22

hocaltar, 2023-02-19:
### Question

Gents, 

I have one bug left and it's tormenting me. 

Has anyone figured out how to maintain recording while switching between front and back cameras? 

Please help!?

### What I tried

I thought I had it pinned down. But for some reason it will not start recording again after I pause while switching from back to front.

### VisionCamera Version

2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-02-22:
On iOS this works flawlessly, on Android this is a limitation in the native CameraX library. I can't do anything to avoid that.

hocaltar, 2023-02-22:
Thank you for the repo and response! 

Follow up question: with V3 utilizing Camera2 instead of CameraX, will pause and resume work on android? 

Thanks again!

ratz6, 2023-04-18:
damn it's been 1 year already for this issue ?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
Flipping Camera device now works on Android!

mrousavy, 2023-09-30:
and it was really hard to implement lol... I had to create my own custom OpenGL pipeline (400 lines of C++ code) just for that.

closed 2023-09-30 by mrousavy

rajyuva1000, 2023-11-20:
hi, could you please provide Flipping Camera device document