margelo/react-native-vision-camera #1489: 🐛 no able to build Android

janet-rivas, 2023-02-21:
### What were you trying to do?

I'm trying to build Android using react-native run-android but instead I'm getting this error  Task :react-native-vision-camera:downloadBoost FAILED 
javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

### Reproduceable Code

```tsx
npx react-native run-android
```

### What happened instead?

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:downloadBoost'.
> javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

### Relevant log output

```shell
> Task :react-native-vision-camera:downloadBoost FAILED
Download https://github.com/react-native-community/boost-for-react-native/releases/download/v1.63.0-0/boost_1_63_0.tar.gz
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

OS info
System:
    OS: macOS 12.6
    CPU: (16) x64 Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz
    Memory: 78.82 MB / 16.00 GB
    Shell: 5.8.1 - /bin/zsh
  Binaries:
    Node: 16.16.0 - /usr/local/bin/node
    Yarn: 1.22.19 - /usr/local/bin/yarn
    npm: 8.11.0 - /usr/local/bin/npm
    Watchman: 2022.08.08.00 - /usr/local/bin/watchman
  Managers:
    CocoaPods: 1.11.3 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.2, iOS 16.2, macOS 13.1, tvOS 16.1, watchOS 9.1
    Android SDK:
      API Levels: 21, 23, 27, 28, 29, 30, 31, 32, 33
      Build Tools: 28.0.3, 29.0.0, 29.0.2, 29.0.3, 30.0.2, 30.0.3, 31.0.0, 32.0.0, 33.0.0, 33.0.1, 33.0.2
      System Images: android-23 | Google APIs ARM EABI v7a, android-23 | Google APIs Intel x86 Atom, android-23 | Google APIs Intel x86 Atom_64, android-26 | Intel x86 Atom_64, android-26 | Google APIs Intel x86 Atom, android-26 | Google APIs Intel x86 Atom_64, android-27 | Intel x86 Atom_64, android-27 | Google APIs Intel x86 Atom, android-28 | Intel x86 Atom, android-28 | Intel x86 Atom_64, android-28 | Google APIs Intel x86 Atom, android-28 | Google APIs Intel x86 Atom_64, android-28 | Google Play Intel x86 Atom, android-28 | Google Play Intel x86 Atom_64, android-28 | Google X86_ARM Intel x86 Atom, android-29 | Google APIs Intel x86 Atom_64, android-30 | Google APIs Intel x86 Atom_64, android-31 | Google APIs Intel x86 Atom_64, android-32 | Google APIs Intel x86 Atom_64, android-33-ext4 | Google Play Intel x86 Atom_64, android-33 | Google APIs ARM 64 v8a, android-33 | Google APIs Intel x86 Atom_64
      Android NDK: Not Found
  IDEs:
    Android Studio: 2022.1 AI-221.6008.13.2211.9514443
    Xcode: 14.2/14C18 - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.11 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 17.0.2 => 17.0.2 
    react-native: 0.67.4 => 0.67.4 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Device

Any Android Simulator 

### VisionCamera Version

2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

KoalaSat, 2023-02-25:
I have exactly the same issue

RodSarhan, 2023-03-02:
I'm using this patch to download boost from another site for the time being, but not sure how reliable it is for the future
version:```2.15.4```
file:```/patches/react-native-vision-camera+2.15.4.patch```
content:
```
diff --git a/node_modules/react-native-vision-camera/android/build.gradle b/node_modules/react-native-vision-camera/android/build.gradle
index 91e64d9..85af638 100644
--- a/node_modules/react-native-vision-camera/android/build.gradle
+++ b/node_modules/react-native-vision-camera/android/build.gradle
@@ -353,7 +353,7 @@ if (ENABLE_FRAME_PROCESSORS) {
 
   task downloadBoost(dependsOn: createNativeDepsDirectories, type: Download) {
     def transformedVersion = BOOST_VERSION.replace("_", ".")
-    def srcUrl = "https://boostorg.jfrog.io/artifactory/main/release/${transformedVersion}/source/boost_${BOOST_VERSION}.tar.gz"
+    def srcUrl = "https://jaist.dl.sourceforge.net/project/boost/boost/${transformedVersion}/boost_${BOOST_VERSION}.tar.gz"
     if (REACT_NATIVE_VERSION < 69) {
       srcUrl = "https://github.com/react-native-community/boost-for-react-native/releases/download/v${transformedVersion}-0/boost_${BOOST_VERSION}.tar.gz"
     }

```

waleedshkt, 2023-05-15:
This is a downloading issue. 

Nothing wrong with the code. Rather it looks a server issue. Either there is an API throttling owing to too many download requests or the server is not responding to enable download. I my case, it downloaded till 50KB/125MB and then failed.

vikingprank, 2023-06-27:
Had `Execution failed for task ':react-native-vision-camera:downloadBoost'.`
Fixed with @RodSarhan patch, cheers!
Download of `:react-native-vision-camera:downloadBoost` is just much longer now

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy