margelo/react-native-vision-camera #1490: ❓

syblus446, 2023-02-22:
### Question

how to take photos of a particular size using react native vision camera? i want to take photos of a particular size how can I do that in react native cli? 

### What I tried

I tried adding a style prop to the camera but it didn't work like that 

```
export default function Camera({ navigation }) {
  const [cameraPermission, setCameraPermission] = useState();

  useEffect(() => {
    (async () => {
      const cameraPermissionStatus = await Camera.requestCameraPermission();
      setCameraPermission(cameraPermissionStatus);
    })();
  }, []);

  const devices = useCameraDevices();
  const cameraDevice = devices.back;
  const cameraRef = useRef();

  return (
    <View style={{ flex: 1, backgroundColor: 'black' }}>
      {
        cameraDevice && cameraPermission === 'authorized' ?
          <Camera
            ref={cameraRef}
            style={styles.camera}
            device={cameraDevice}
            isActive={true}
            photo={true}
            focusable={true}
            fps={60}
            frameProcessorFps={1}
            preset={true}
            enableZoomGesture={true}
            hdr={true}
            AutoFocusSystem="phase-detection"
            qualityPrioritization="quality"
          />
          :
          <ActivityIndicator color='white' size='large' />
      }
    </View>
  );
}

const styles = StyleSheet.create({
  camera: {
      borderTopLeftRadius: 30,
      borderTopRightRadius: 30,
      borderBottomLeftRadius: 30,
      borderBottomRightRadius: 30,
      overflow: 'hidden',
      width: '130%',
      aspectRatio: 1
  },
});
```

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-02-23:
That's not how VisionCamera works. Read the docs on selecting Camera Devices and Formats. Then select a format with you target size.

closed 2023-02-23 by mrousavy

syblus446, 2023-02-23:
@mrousavy I looked that the docs but I know understand how to get 2448 x 3264 pixels camera format like everyone can only take photo in 2448 x 3264 pixels How can I do that? Can you give code examples ?