margelo/react-native-vision-camera #1499: Video Recording on iOS has 1-2 seconds cut off at end of video🐛

jkerb8, 2023-02-27:
### What were you trying to do?

Recording a video on an iOS device, with the expectation that the final video file would include the frames from the moment "startRecording" was called until the moment "stopRecording" was called.

### Reproduceable Code

```tsx
<Camera
									ref={ref => {
										this.camera = ref;
									}}
									style={Style.cameraContainer}
									isActive={true}
									device={this.state.cameraDevice}
									audio={true}
									video={true}
									zoom={this.state.zoom}
									videoStabilizationMode={'auto'}
									preset={'hd-1920x1080'}
								></Camera>
```

### What happened instead?

The last 1-2 seconds of video are not present in the final recording file. This problem does not apply to Android devices. I recorded a timer and stopped the video when the timer reached 5 seconds. The video file that was saved ended before the timer reached 4 seconds.

### Device

iPad 9th Gen (15.5)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

jkerb8, 2023-11-03:
I'm unable to upgrade to VisionCamera V3 until the orientation issues are resolved, and this problem still occurs with the latest version of V2. In addition, the audio is out of sync with the video, as it seems like the video recording starts 1-2 seconds before the button is pressed, and ends 1-2 seconds early. The audio is actually correct but since the video is not, it appears out of sync. I can also confirm this happens with iOS 17 on a variety of iOS devices.

mrousavy, 2023-11-07:
>  In addition, the audio is out of sync with the video

In V3?

jkerb8, 2023-11-07:
> > In addition, the audio is out of sync with the video
> 
> In V3?

This only happens in V2. Apologies if I'm not supposed to open issues for V2 anymore, I know there's only limited support being provided.

mrousavy, 2023-11-09:
Ah - yea no worries mate! But yes you're right I can't really provide support for both codebases. Hope you'll figure this one out or are able to upgrade to V3!