margelo/react-native-vision-camera #1502: How to detect Huaman face using react-native-vision-camera?

syblus446, 2023-03-01:
### Question

How to detect a human face using react-native-vision-camera? how to do things in real-time? i am using react native cli I also installed react-native-reanimated and @tensorflow-models/blazeface and @tensorflow/tfjs how can I detect a human face? i can also use any other package also can anyone can give me code examples to do that? 

### What I tried

import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { TouchableOpacity, View, StyleSheet, Platform, ActivityIndicator } from 'react-native';
import { useState, useEffect, useRef } from "react";
import * as tf from '@tensorflow/tfjs';
import * as blazeface from '@tensorflow-models/blazeface';
import 'react-native-reanimated'

export default function Camera({ navigation }) {

  const [cameraPermission, setCameraPermission] = useState();
  const [model, setModel] = useState(null);
  const [faces, setFaces] = useState([]);

  useEffect(() => {
    (async () => {
      const cameraPermissionStatus = await Camera.requestCameraPermission();
      setCameraPermission(cameraPermissionStatus);
    })();
  }, []);

  useEffect(() => {
    async function loadModel() {
      const model = await blazeface.load();
      setModel(model);
    }
    loadModel();
  }, []);

  const devices = useCameraDevices('wide-angle-camera')
  const cameraDevice = devices.front;
  const cameraRef = useRef();

  const handleTakePicture = async () => {
    const photo = await cameraRef.current.takePhoto();
    console.log('photo size is', photo)
    const path = Platform.OS === 'android' ? 'file://' + photo.path : photo.path;
  }

  const detectHumanFaces = async (frame) => {
    if (!model) {
      return;
    }
    const options = { maxFaces: 1, scoreThreshold: 0.5 };
    const returnTensors = false;
    const predictions = await model.estimateFaces(
      frame,
      returnTensors,
      options,
    );
    if (predictions.length === 0) {
      console.log("No face detected");
    } else {
      const isHumanFace = predictions.some(prediction => {
        return prediction.class === "face" && prediction.subclass === "human";
      });
      if (isHumanFace) {
        console.log("Human face detected");
      }
      setFaces(predictions);
    }
  }

  return (
    <View style={{ flex: 1, backgroundColor: 'black' }}>
      {
        cameraDevice && cameraPermission === 'authorized' ?
          <Camera
            ref={cameraRef}
            style={styles.camera}
            device={cameraDevice}
            isActive={true}
            photo={true}
            focusable={true}
            fps={240}
            frameProcessor={async frame => {
              if (!frame || !model) {
                return;
              }
              const options = { maxFaces: 1, scoreThreshold: 0.5 };
              const returnTensors = false;
              const predictions = await model.estimateFaces(
                frame,
                returnTensors,
                options,
              );
              if (predictions.length === 0) {
                console.log("No face detected");
              } else {
                setFaces(predictions);
              }
            }}
            frameProcessorFps='auto'
            enableHighQualityPhotos={true}
          />
          :
          <ActivityIndicator color='white' size='large' />
      }
      <TouchableOpacity
        style={{
          alignSelf: 'center',
          alignItems: 'center',
          width: 90,
          height: 90,
          borderRadius: 500,
          marginTop: '15%',
          borderColor: 'white',
          borderWidth: 6,
          marginBottom: 30
        }}
        onPress={handleTakePicture}
      >
        <View style={{ opacity: 0.5 }} />
      </TouchableOpacity>
    </View>
  );
}

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

closed 2023-03-02 by syblus446

CodingWithNobody, 2023-05-01:
Try this https://www.youtube.com/watch?v=mv4c2uTxR4U

mohemos, 2024-07-09:
@xts-bit are you able to solve this issue? Can you dectect human face using React native vision camera ?