margelo/react-native-vision-camera #1503: No member named 'ShareableValue' in namespace 'reanimated'🐛

Romick2005, 2023-03-01:
### What were you trying to do?

Build iOS project with react-native-reanimated v3.0.0 (pod 1.11.3).
[image]

### Reproduceable Code

```tsx
npm i react-native-reanimated@latest && cd ios && pod install
```

### What happened instead?

Build failed.

### Relevant log output

```shell
/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorRuntimeManager.mm:146:32 No member named 'ShareableValue' in namespace 'reanimated'
```

### Device

iPhone XS

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

idrissakhi, 2023-03-03:
Actually having the same thing

st0ffern, 2023-03-04:
@idrissakhi thanks!
Downgrade `react-native-reanimated` to `2.14.4` and run `pod install`

alexstanbury, 2023-03-06:
v3 of this library removes the dependency on reanimated, so if you want to use reanimated v3 then you'll need that.

idrissakhi, 2023-03-06:
@alexstanbury v3 is in release candidates, so for me, it's not ready yet for implementation

alexstanbury, 2023-03-06:
@idrissakhi I understand that, just advising the situation. 

Be aware that v2 of reanimated has a lot of memory related issues, so if your app needs to remain open for long periods of time like mine does then your users are going to feel the effects of the memory leaks.

albindb92, 2023-03-09:
Hi, any fix available for react-native-reanimated v3 without downgrading?

TomWq, 2023-03-14:
When can react-native-reanimated v3 be compatible

deflexable, 2023-03-18:
has anyone been able to fix this without downgrading rn-reanimated?

bastiaanv, 2023-03-23:
v3 doesn't need reanimated. So we need to wait for that I guess?

https://github.com/mrousavy/react-native-vision-camera/issues/1376

miguel-martinez-6661, 2023-03-24:
Any news about this issue? Do we need to wait for the v3?

Ahmedhamed77, 2023-04-20:
I am facing the same problem with react-native 0.71.6 any solution without downgrading reanimated?

bastiaanv, 2023-04-20:
We switched to [react-native-camera-kit](https://github.com/teslamotors/react-native-camera-kit). That one does not use reanimated

kshanmugam73, 2023-05-18:
I encountered the same issue with version 0.71.4, but after updating to `"react-native-reanimated": "^2.3.0"` in my project, I received a positive response.

Jad-Jbara, 2023-06-02:
As of reanimated 3, the reanimated 1 APIs are dropped. This library needs to be updated to support reanimated 3

Also, one of the maintainers stated that reanimated 3 isn't supported yet by  `react-native-vision-camera`. [issue](https://github.com/software-mansion/react-native-reanimated/issues/3831#issuecomment-1359263674)

Romick2005, 2023-06-26:
> Actually having the same thing. it's only related to reanimated 3, if you downgrade then it will work just fine

And now [react-native-reanimated@2.17.0 is not supported by react-native@0.72.0](https://github.com/software-mansion/react-native-reanimated/issues/4522). 
So your downgrade is very bad idea. 

So this one is probably better solution to not have modules that that use internal code of other modules or suggest to add unpublished [react-native-worklets and @shopify/react-native-skia](https://github.com/mrousavy/react-native-vision-camera/pull/1466#issuecomment-1553381626) that you have to install separate as react-native-vision-camera does not have them as internal dependancy.

So +1 for [react-native-camera-kit](https://github.com/teslamotors/react-native-camera-kit)

Romick2005, 2023-07-05:
If you do not need frame processor you can patch react-native-vision-camera in order to build on iOS with latest react-native-reanimated@3.3.0.
```
--- a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorRuntimeManager.mm	
+++ b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorRuntimeManager.mm	
@@ -143,7 +143,7 @@ - (void) installFrameProcessorBindings {
 
     auto viewTag = arguments[0].asNumber();
     NSLog(@"FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...");
-    auto worklet = reanimated::ShareableValue::adapt(runtime, arguments[1], runtimeManager.get());
+    // auto worklet = reanimated::ShareableValue::adapt(runtime, arguments[1], runtimeManager.get());
     NSLog(@"FrameProcessorBindings: Successfully created worklet!");
 
     RCTExecuteOnMainQueue([=]() {
@@ -155,9 +155,9 @@ - (void) installFrameProcessorBindings {
         NSLog(@"FrameProcessorBindings: Converting worklet to Objective-C callback...");
 
         auto& rt = *runtimeManager->runtime;
-        auto function = worklet->getValue(rt).asObject(rt).asFunction(rt);
+        // auto function = worklet->getValue(rt).asObject(rt).asFunction(rt);
 
-        view.frameProcessorCallback = convertJSIFunctionToFrameProcessorCallback(rt, function);
+        // view.frameProcessorCallback = convertJSIFunctionToFrameProcessorCallback(rt, function);
         NSLog(@"FrameProcessorBindings: Frame processor set!");
       });
     });

```

sravis02, 2023-07-28:
@Romick2005 Thank you a lot mate, this works for IOS! 
Do you also have a patch for Android ? :))

Romick2005, 2023-07-28:
It works for Android as is, but not with the latest reanimation 3.4.0

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

PR #1759 (2023-09-18): feat: Reanimated v3 support in RNVisionCamera v2
## What

**This PR makes it possible to use RNVisionCamera v2 with Reanimated v3 (3.5.0+)** 🥳 

I've branched off from @mrousavy's PR (https://github.com/mrousavy/react-native-vision-camera/pull/1623) which bumps react-native version in the example app as well as does some general cleanup and integrated Reanimated v3 on top of it.

## Changes

* Move Camera thread runtime creation from C++ to JS using `Reanimated.createWorkletRuntime`
* Call worklets using `WorkletRuntime::runGuarded` which already supports error handling
* Consume Reanimated as [... cut]
closed 2023-09-30 by mrousavy