margelo/react-native-vision-camera #1504: How to detect a face using react-native-vision-camera?

syblus446, 2023-03-02:
### Question

How to detect a human face using react-native-vision-camera? how to do things in real-time? I tried to use vision-camera-face-detector but it gave me Frame Processor threw an error: Can't find variable: __scanFaces error also it does not support Hermes Then how to use react-native-vision-camera Frame Processor to detect human face? Is it possible that any contributor can add a pre-built prop to detect human faces using a vision-camera-face-detector as its a Community Plugin?

### What I tried

import * as React from 'react';
import { runOnJS } from 'react-native-reanimated';

import { StyleSheet } from 'react-native';
import {
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';

import { Camera } from 'react-native-vision-camera';
import { scanFaces, Face } from 'vision-camera-face-detector';

export default function App() {
  const [hasPermission, setHasPermission] = React.useState(false);
  const [faces, setFaces] = React.useState<Face[]>();

  const devices = useCameraDevices();
  const device = devices.front;

  React.useEffect(() => {
    console.log(faces);
  }, [faces]);

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const scannedFaces = scanFaces(frame);
    runOnJS(setFaces)(scannedFaces);
  }, []);

  return device != null && hasPermission ? (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      frameProcessor={frameProcessor}
      frameProcessorFps={5}
    />
  ) : null;
}

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

EkoSetiyo13, 2023-03-05:
edit file babel.config.js like this

      module.exports = {
        presets: ['module:metro-react-native-babel-preset'],
        plugins: [
          [
            'react-native-reanimated/plugin',
            {
              globals: ['__scanFaces'],
            },
          ],
        ],
      };

closed 2023-03-13 by mrousavy

CodingWithNobody, 2023-05-01:
Try this https://www.youtube.com/watch?v=mv4c2uTxR4U