margelo/react-native-vision-camera #1510: 🐛 Property '_setGlobalConsole' doesn't exist (react-native-reanimated@^3)

chrisribal, 2023-03-03:
### What were you trying to do?

I tried to reproduce the basic "image labeler" example to use a frame processor from the documentation.

### Reproduceable Code

```tsx
import React, { useState, useEffect } from 'react';
import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { View, Text } from 'react-native';
import { labelImage } from 'vision-camera-image-labeler';

const CameraTest = (props) => {
    const devices = useCameraDevices();
    const device = devices.back;

    const frameProcessor = useFrameProcessor((frame) => {
       'worklet';
        const labels = labelImage(frame)
        console.log(labels);
    }, [])

    useEffect(() => {
        const permitCamera = async () => {
            const cameraPermission = await Camera.getCameraPermissionStatus()

            if (cameraPermission !== 'authorized') {
                await Camera.requestCameraPermission()
            }
        }

        permitCamera();
    }, []);

    if (!device) {
        return <View>
            <Text>Loading...</Text>
        </View>
    }

    return (
        <View>
            <Camera
                device={device}
                style={{
                    width: 500,
                    height: 500
                }}
                frameProcessor={frameProcessor}
                frameProcessorFps={5}
                isActive={true}
            />            
        </View>
    )
};

export default CameraTest;
```

### What happened instead?

The App crashes with an error due to non existent property '_setGlobalConsole'. This is due to changes in react-native-reanimated@^3. I tried using react-native-reanimated@^2 instead, but v2 is not working with react-native@^0.71 used in my project.

My current config:
- react-native@0.71.3
- react-native-reanimated@^3.0.0
- react-native-vison-camera@2.15.4
- vision-camera-image-labeler@0.1.6

### Device

Xiaomi Redmi Note 10 Pro

### VisionCamera Version

^2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Mistifiou, 2023-03-06:
Hi

Based on [this pull request](https://github.com/software-mansion/react-native-reanimated/pull/3853)

Temporary adding this to index.js file seems to solve issue for me:

```
import {AppRegistry} from 'react-native';

/**
 * @see https://github.com/mrousavy/react-native-vision-camera/issues/1510
 * @see https://github.com/software-mansion/react-native-reanimated/pull/3853
 */
global._setGlobalConsole = (_val) => {
    // noop
  };
  
import 'react-native-reanimated';
[...]
```

zzz08900, 2023-04-01:
react-native-reanimated 2.x does work with react-native 0.71 using all default settings. If that does not work you then there's likely something special about your specific project and you'll have to find a solution yourself.

yormy, 2023-04-05:
It seems not to work with react-native-reanimated v3

yormy, 2023-04-05:
@chrisribal were you able to solve or find a workaround ?

chrisribal, 2023-04-05:
> @chrisribal were you able to solve or find a workaround ?

No, I ended up downgrading REA, as there were more incompatibilities. I guess we should wait for v3 to be ready, which also seems to support REA 3.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

PR #1759 (2023-09-18): feat: Reanimated v3 support in RNVisionCamera v2
## What

**This PR makes it possible to use RNVisionCamera v2 with Reanimated v3 (3.5.0+)** 🥳 

I've branched off from @mrousavy's PR (https://github.com/mrousavy/react-native-vision-camera/pull/1623) which bumps react-native version in the example app as well as does some general cleanup and integrated Reanimated v3 on top of it.

## Changes

* Move Camera thread runtime creation from C++ to JS using `Reanimated.createWorkletRuntime`
* Call worklets using `WorkletRuntime::runGuarded` which already supports error handling
* Consume Reanimated as [... cut]
closed 2023-09-30 by mrousavy