margelo/react-native-vision-camera #1511: Why ultra-wide-angle-camera is not working ? and wide-angle-camera is working?

syblus446, 2023-03-03:
### Question

Why ultra-wide-angle-camera is not working? and wide-angle-camera is working? I am trying to take very high-quality photos with my front camera but when i change my CameraDevices to ultra-wide-angle-camera it's not open the camera. but the wide-angle-camera is working and opening the camera how can i use ultra-wide-angle-camera in front camera? 

Also What is Frontside Wide-Angle Camera (FaceTime HD)? how do enable that? 

### What I tried

```

  const devices = useCameraDevices('ultra-wide-angle-camera ');
  const cameraDevice = devices.front;
  const cameraRef = useRef();
```

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Naruto0829, 2023-07-09:
Hi, I have one question. I am going to use react-native-vision-camera in the expo way.
But I faced with some errors. I just wanna know if it is possible to use ultra-wide camera in expo.
I will be happy If someone know it is possible or not in expo.
Thanks

tvpratik, 2023-08-25:
format comment and try to use camera

tvpratik, 2023-08-25:
@xts-bit 
camera using this code 
  const supportsCameraFlipping = useMemo(() => devices.back != null && devices.front != null, [devices.back, devices.front]);

xts-bit, 2023-08-25:
@tvpratik What do you mean?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

algopert, 2023-10-11:
Hi,

I hope you're doing well. It seems that your current library is unable to detect the ultra-wide-angle camera on Google Pixel 6 Pro and Google Pixel 7 Pro, despite their inclusion of this camera. 
This issue persists across the entire Google phone series, and I've tried both versions 2 and 3 of the library without success. 
Could you please provide some insight into why this detection is currently impossible?

Thank you for your assistance.

mrousavy, 2023-10-12:
@algopert you already created an issue for this, stop spamming.

Anyways, here's the answer I already gave you before:

VisionCamera has a very solid Devices API that is capable of detecting all Camera devices on your phone that are available for use in your app. Even USB Cameras. If there's a device missing in that list, it is pretty likely that this device is simply not available for use, and the vendor (Samsung, Huawei, ...) just locked this and doesn't allow you to use this specific Camera device. In other words; if VisionCamera can't detect the Camera, there's nothing you can do about it. It's not even available in a native Android app (except for their stock Samsung/Huawei/... Camera app).

To confirm this, download the official Google example for Camera apps; [camera-samples/Camera2Video](https://github.com/android/camera-samples/tree/main/Camera2Video) and run it on your phone. If the Camera device is not in there either, then again, nothing you can do.

I'm sure that for specific devices such as telephoto or ultra-wide, they usually don't allow you to use those devices directly, but only in combination with other devices (e.g. a "dual-camera" consisting of a normal wide-angle and the ultra-wide-angle), because they only work like that. 

Try logging all devices you get:

```js
const devices = Camera.getAvailableCameraDevices()
devices.forEach((d) => {
  d.formats = null
})
console.log(JSON.stringify(devices, null, 2))
```

And maybe there's a multi-cam device that has the ultra-wide-angle camera in there. (Try setting `zoom` to the minimum available value (`device.minZoom`) and see if it is zoomed out or not)

tvpratik, 2023-10-18:
> Hi,
>
> I hope you're doing well. It seems that your current library is unable to detect the ultra-wide-angle camera on Google Pixel 6 Pro and Google Pixel 7 Pro, despite their inclusion of this [...]

it is possible

algopert, 2023-10-18:
> it is possible

I checked, but impossible.
Could you please help me?