margelo/react-native-vision-camera #1514: ERROR  ReferenceError: Property '_setGlobalConsole' doesn't exist

syblus446, 2023-03-06:
### Question

I am facing a error that error says  ERROR  ReferenceError: Property '_setGlobalConsole' doesn't exist any idea how can i fix this issue with react-native-vision-camera?  

### What I tried

```
 "dependencies": {
    "@react-native-async-storage/async-storage": "^1.17.11",
    "@react-navigation/native": "^6.1.6",
    "@react-navigation/native-stack": "^6.9.12",
    "firebase": "^9.17.2",
    "react": "18.2.0",
    "react-native": "0.71.3",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-image-crop-picker": "^0.39.0",
    "react-native-onboarding-swiper": "^1.2.0",
    "react-native-reanimated": "^3.0.1",
    "react-native-safe-area-context": "^4.5.0",
    "react-native-screens": "^3.20.0",
    "react-native-vector-icons": "^9.2.0",
    "react-native-vision-camera": "^2.15.4",
    "vision-camera-face-detector": "^0.1.8"
  },
```

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

GaylordP, 2023-03-10:
Same problem for a week :(
Did you manage to solve the problem @syblus446 ?

GaylordP, 2023-03-10:
Thank you for your answer, the error is still there for me after this addition :/

socrates93, 2023-03-10:
I have the same situation. Is there another version which does not have this bug? I've tested this library a couple of days ago but didn't have this issue. My mistake removing that test project :(

Taraslysak, 2023-03-11:
From our investigation, `useFrameProcessor` hook uses `_setGlobalConsole` global method from `react-native-reanimated` library, but it was removed in the latest version (3.0.2) of `react-native-reanimated`.

syblus446, 2023-03-11:
using react-native-reanimated version 2.14.4 fixed my issue

niubiletei, 2023-03-11:
I have the same situation，and I got a location on the relevant file:`node_modules\react-native-vision-camera\src\hooks\useFrameProcessor.ts`  
[image]
but when I comment out this line of code,the program will throw an error and crashes  
[image]
I don't know how to solve it

socrates93, 2023-03-12:
> using react-native-reanimated version 2.14.4 fixed my issue

Thanks, @syblus446 - I can confirm that it is working with `react-native-reanimated@2.14.4`.

niubiletei, 2023-03-15:
hi,guys.if you can't resolve this question,you should try this https://github.com/react-native-camera/react-native-camera/tree/master

rizkhal, 2023-03-17:
I'm trying to downgrade using `react-native-reanimated@2.14.4` but my app crashes

Btw i use: 
```json
"react-native": "0.71.4",
"react": "18.2.0",
```

socrates93, 2023-03-17:
> I'm trying to downgrade using `react-native-reanimated@2.14.4` but my app crashes
> 
> Btw i use:
> 
> ```json
> "react-native": "0.71.4",
> "react": "18.2.0",
> ```

Try running `react-native start --reset-cache`. I use the following settings, consider a downgrade to this:
```json
     "react": "18.1.0",
    "react-native": "0.70.6",
```

rizkhal, 2023-03-17:
> > I'm trying to downgrade using `react-native-reanimated@2.14.4` but my app crashes
> > Btw i use:
> > ```json
> > "react-native": "0.71.4",
> > "react": "18.2.0",
> > ```
> 
> Try running `react-native start --reset-cache` and clean other cache on android/ios folders. I use the following settings, consider a downgrade to this:
> 
> ```json
>     "react": "18.1.0",
>     "react-native": "0.70.6",
> ```

I have a lot of error after trying to downgrade react & react native like your suggestion

socrates93, 2023-03-17:
> I have a lot of error after trying to downgrade react & react native like your suggestion

Ugh. Sorry. Did you clean your cache? Try this:
```json
 watchman watch-del-all && rm -rf $TMPDIR/react-native-packager-cache-* &&
 rm -rf $TMPDIR/metro-bundler-cache-* && rm -rf node_modules/ && npm cache clean --force &&
 npm install && npm start -- --reset-cache
```

R4tch3t, 2023-03-23:
https://github.com/mrousavy/react-native-vision-camera/pull/1466

ivaldo, 2023-03-30:
> > using react-native-reanimated version 2.14.4 fixed my issue
> 
> Thanks, @syblus446 - I can confirm that it is working with `react-native-reanimated@2.14.4`.

It's work for me after 'npx react-native start --reset-cache'
```
    "react": "17.0.2",
    "react-native": "0.68.2",
    "react-native-reanimated": "2.14.4",
    "react-native-vision-camera": "2.15.4",
    "vision-camera-code-scanner": "0.2.0"
```

Guilherme03, 2023-04-01:
> 

Thanks a lot. Everything ok.

ustuncem, 2023-04-10:
This issue still persists. I am using reanimated v3.0.2 . It appears that reanimated team has removed _setGlobalConsole. https://github.com/software-mansion/react-native-reanimated/pull/3853/files

GSFZamai, 2023-04-13:
https://github.com/mrousavy/react-native-vision-camera/issues/1210#issuecomment-1457776475

vision-camera V2 only works with reanimated v2.
On vision-camera v3 it will not use reanimated anymore and the incompatibility will be resolved.
So until the release we should use reanimated v2

Tarkeshwar108, 2023-05-02:
I was also facing this issue but it resolved with below step:
 **step1:** I downgraded "react-native-reanimated": "^3.1.0"  to  "react-native-reanimated": "2.14.4".
 **step2:**  Added `import 'react-native-reanimated'` in index.js which is our entry file.
 **step3**:  `yarn start --reset-cache`
After above step run your project. It should run without any issue.
**My packages details:**
     "react": "18.2.0",
    "react-native": "0.71.3",
    "react-native-reanimated": "2.14.4",
    "react-native-vision-camera": "2.15.4",
    "vision-camera-code-scanner": "0.2.0"

maksym-bezditko, 2023-05-03:
[Tarkeshwar108](https://github.com/Tarkeshwar108)

Thanks, that worked for me with "react": "17.0.2", "react-native": "0.68.0", but I noticed a significant bug in react-native-vision-camera of > 2.13.0 version. If I go to the next screen which uses Camera component from react-native-vision-camera, the component displays a part of previous screen for a second (when it is mounting as far as I understand). Same when that next screen goes out of focus. With react-native-vision-camera 2.13.0 nothing of that kind was observed. Any ideas how to workaround it?

maksym-bezditko, 2023-05-03:
And can't build with react-native-animated version you mentioned and react-native-vision-camera >= 2.15.0. Maybe you know react-native-reanimated version that is compatible with react-native-vision-camera 2.13.0. Thank you in advance!

gift-sequoiaat, 2023-07-06:
> using react-native-reanimated version 2.14.4 fixed my issue

> Ugh. Sorry. Did you clean your cache? Try this:
>
> ```json
>  watchman watch-del-all && rm -rf $TMPDIR/react-native-packager-cache-* &&
>  rm -rf $TMPDIR/metro-bundler-cache-* && rm -rf node_modules/ && [...]

use react-native-reanimated version 2.17 that should fix the issue

hernanif1, 2023-07-10:
trying to downgrade reanimated gives me this error:
`Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.`

elibroftw, 2023-08-02:
When I set the version of reanimated to 2.14.4, I get the issue

```
eerror Failed to install the app. Command failed with exit code 1: gradlew.bat tasks FAILURE: Build failed with an exception. * Where: Build file 'C:\Users\maste\Documents\GitHub\SplitTheTank\node_modules\react-native-reanimated\android\build.gradle' line: 292 * What went wrong: A problem occurred evaluating project ':react-native-reanimated'. > Could not get unknown property 'minor' for project ':react-native-reanimated' of type org.gradle.api.Project
```

To fix this issue, I have to manually patch node_modules????

Mynameisjohndev, 2023-08-04:
I have the same situation in my application

lib versions
```
    "expo": "~49.0.5",
    "react": "18.2.0",
    "react-native": "0.72.3",
    "vision-camera-code-scanner": "^0.2.0",
    "react-native-vision-camera": "^2.15.4",
    "react-native-reanimated": "^2.17.0"
```
i remove this part
[image]

add this
[image]

upgrade gradle.properties to
[image]

these were the things i did and i got the same error, in case anyone knows how to fix

umusicanumothu, 2023-08-06:
On react native v0.72, the downgrade to 2.14.4, is not working. :( For a quick fix, I have made a change in node_module react -native-vision-camera the below changes. 

1. go to node_modules->react-native-vision-camera->src->hooks->useFrameProcessor.ts
2. comment the code `_setGlobalConsole(console);`
3. restart the metro with reset cache. `react-native start --reset-cache`

Its just a quick fix. Everytime run yarn command, it will be replaced. For persistant change, please apply `patch-package`
https://www.npmjs.com/package/patch-package

Mynameisjohndev, 2023-08-07:
If I install the patch-package, I make the change inside the library installed in my application, I run the command npx patch-package react-native-vision-camera? @umusicanumothu

truongto, 2023-08-07:
> On react native v0.72, the downgrade to 2.14.4, is not working. :( For a quick fix, I have made a change in node_module react -native-vision-camera the below changes.
>
> 1. go to [...]
----------------------

I got an error :
 ERROR  frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting, js engine: hermes

Mynameisjohndev, 2023-08-07:
I still haven't tested this way as he said, I used expo's own library to expo-barcode-scanner to solve the temporary problem, probably in the future in version 3 of this library that will no longer use react-native-reanimated I'll come back to use her
 @truongto

VivekCEpic, 2023-08-08:
@mrousavy I have the same issue with my project.

Please anyone help with this? Thank you in advance.

` "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-reanimated": "3.0.2"
    "react-native-vision-camera": "^2.15.4",`

Dhiraj-j, 2023-08-16:
Anyone able to solve the issues 

"react": "18.2.0",
"react-native": "0.72.3",
"react-native-vision-camera": "^2.15.4",
"react-native-reanimated": "3.3.0",
"vision-camera-code-scanner": "^0.2.0"

Antsthebul, 2023-08-22:
FWIW, downgrading worked for me. I was using react-native 72 but it wasnt working. Aside from the warnings I get, it seems to work. I had to install the package that npm i react-native-reanimated says to remove...
react 18.2
react-native 71.3
react native renimated 2.14.4
react-native-community/cli-platform-ios 11.3.7
react-native-vision-camera 2..15.4

Also, if your downgrading react-native version like me the Podfile needs require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules' at the top.

itzFaiz, 2023-08-25:
Did anyone figure out how to solve the issue. I tried downgrading as per above suggestions but getting too many errors. please help i have been stuck on it from 2 weeks.

pavalappdev, 2023-09-08:
// @ts-expect-error _setGlobalConsole is set by RuntimeDecorator::decorateRuntime
    //  _setGlobalConsole(console);
      // @ts-expect-error
      global.didSetConsole = false;
      
      just put this and error will be resolved

pavalappdev, 2023-09-08:
`  // @ts-expect-error _setGlobalConsole is set by RuntimeDecorator::decorateRuntime
    //  _setGlobalConsole(console);
      // @ts-expect-error
      global.didSetConsole = false;`

pavalappdev, 2023-09-08:
`
    //  _setGlobalConsole(console); //comment 
      global.didSetConsole = false;` //make false

abdymm, 2023-09-21:
still having this issue, are you guys able to resolve this?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

PR #1759 (2023-09-18): feat: Reanimated v3 support in RNVisionCamera v2
## What

**This PR makes it possible to use RNVisionCamera v2 with Reanimated v3 (3.5.0+)** 🥳 

I've branched off from @mrousavy's PR (https://github.com/mrousavy/react-native-vision-camera/pull/1623) which bumps react-native version in the example app as well as does some general cleanup and integrated Reanimated v3 on top of it.

## Changes

* Move Camera thread runtime creation from C++ to JS using `Reanimated.createWorkletRuntime`
* Call worklets using `WorkletRuntime::runGuarded` which already supports error handling
* Consume Reanimated as [... cut]
closed 2023-09-30 by mrousavy