margelo/react-native-vision-camera #1516: "auto" flash does not work on samsung devices🐛

MateuszKruk1303, 2023-03-07:
### What were you trying to do?

I wanted to make flash "auto" by default, and as the only option.

### Reproduceable Code

```tsx
const photo = await camera.current.takePhoto({
        flash: "auto",
        enableAutoStabilization: true,
        qualityPrioritization: "quality",
      });

//... saving the photo
```

### What happened instead?

On iOS it works as expected:
- No flash in bright area, flash in pitch black

But on android, there is something wrong:
- No flash in bright area, flash in pitch black - tested on Redmi 9A (Android 10), POCO X3 (Android 12) (Works as Expected)
**- No flash in bright area, no flash in pitch black - tested on 2 Samsung devices (both s10, Android 12)**

I checked also with **on** option, whether flash works

`      const photo = await camera.current.takePhoto({
        flash: "auto",
        enableAutoStabilization: true,
        qualityPrioritization: "quality",
      });
`

and it works, so the problem is with **auto** mode only.

Finally, the problem is:

**_Flash completely does not work in auto mode on samsung devices._**

### Relevant log output

```shell
No logs.
```

### Device

Samsung S10, IPhone 14 Pro, Redmi 9A

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Sewb21, 2023-03-13:
I am also experiencing this

habbababbai, 2023-09-26:
I reproduced the same issue on Samsung Galaxy S7. Probably [this issue](https://github.com/mrousavy/react-native-vision-camera/issues/569) is related and fix got merged to v3 version of app.

mrousavy, 2023-09-30:
Flash currently doesn't work in V3, if I get enough sponsors I'll try implementing it

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Flash in this feature request/issue now: #1890  ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.