margelo/react-native-vision-camera #1519: 🐛  Unable to find a specification for `react-native-worklets` depended upon by `VisionCamera`

idrissakhi, 2023-03-09:
### What were you trying to do?

Integrating the 3.0.0rc2 since lower version depends on react-native-reanimted and SharedValue is not available in new version.

### Reproduceable Code

```tsx
use "react-native-vision-camera": "3.0.0-rc.2" and yarn install && cd && pod install
```

### What happened instead?

Unable to find a specification for `react-native-worklets` depended upon by `VisionCamera`

### Relevant log output

```shell
[!] Unable to find a specification for `react-native-worklets` depended upon by `VisionCamera`

You have either:
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

error: Error: Unable to resolve module react-native-worklets/src from /Users/isakhi/projects/partoo_mobile/node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts: react-native-worklets/src could not be found within the project or in these directories:
  node_modules/react-native-vision-camera/node_modules
  node_modules
  1 | import type { Frame, FrameInternal } from './Frame';
  2 | import { Camera } from './Camera';
> 3 | import { Worklets } from 'react-native-worklets/src';
```

### Device

iphone 14

### VisionCamera Version

3.0.0-rc.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Alexispap, 2023-03-09:
Did you try those commands here: 
yarn add react-native-vision-camera@rc
yarn add react-native-worklets@https://github.com/chrfalch/react-native-worklets#d62d76c
yarn add @shopify/react-native-skia@0.1.175

idrissakhi, 2023-03-09:
nope, for Skia i didn't add it because i am not using frameProcessor. and for react-native-worklets it should be in peerDependencies so users of the lib should not add repo that are not available in npm

mrousavy, 2023-03-13:
> and for react-native-worklets it should be in peerDependencies so users of the lib should not add repo that are not available in npm

That's not really how it works - native modules cannot be detected by React Native that way. So you need to install RN Worklets manually.

idrisssakhi, 2023-03-13:
@mrousavy Thank you for update. what about Skia, is there any way to opt-out from it if we don't use frameProcessor ?

mrousavy, 2023-03-14:
yes there will be.

thohoffm1, 2023-03-27:
It's worth mentioning if you are using an Nx monorepo that to get this to work you need to manually copy the extra deps to the package.json in your RN app root folder.

```
    ...
    "@shopify/react-native-skia": "*",
    "react-native-worklets": "*",
    ...
```

dehimer, 2023-04-28:
Hello! Faced with same problem (also tried to avoid initial problem, same as in author's case). So, is it sorted already? Or may be will sorted out?

ombogdanJoinToIt, 2023-10-17:
same problem

smfunder, 2023-11-01:
I got the same issue. I tried the solutions above and can't get it working. I'm using react-native-vision-camera 3.6.4. Previously I've tried to install react-native-worklets without specifying the repo but later I removed that. Maybe that is causing the issue?

excelso, 2024-01-06:
Add this in your babel.config.js

`module.exports = {
  plugins: [
    ["react-native-worklets-core/plugin"],
    // ...
  ],
  // ...
};`

`npm start --reset-cache`

zjkuang, 2024-01-09:
> Add this in your babel.config.js
> 
> ```
> 
> module.exports = {
>   plugins: [
>     ["react-native-worklets-core/plugin"],
>     // ...
>   ],
>   // ...
> };
> ```
> 
> `pod install` `npm start --reset-cache`

With this added, I am still seeing the same `pod install` failure
```
[!] Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera`
```

dipakatkari, 2024-09-16:
1. `npm install react-native-worklets-core`

2. Add this in your babel.config.js - 
` ['react-native-worklets-core/plugin']`

3. cd ios and -
`pod install`

This works for me