margelo/react-native-vision-camera #1521: 🐛 Unable to start the react-native app after installing react-native-vision-camera.

Szabolcs11, 2023-03-11:
### What were you trying to do?

I simply create a react native app. And after that I install the react-native-vision-camera, and I unable to start the project.

### Reproduceable Code

```tsx
npx react-native@latest init AwesomeProject
npm i react-native-vision-camera
cd AwersomeProject
npm run start
npm run android
```

### What happened instead?

I got an error. 
When I got to ./android and try to run ./gradlew clean I got the same issue.

### Relevant log output

```shell
PS C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp> npm run android

> tsisappp@0.0.1 android  
> react-native run-android

info Starting JS server...
info Installing the app...

> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp\node_modules
VisionCamera: Building with Hermes...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are disabled because REA v2 does not exist. C++ part will not be built.
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.
5 actionable tasks: 5 up-to-date

FAILURE: Build failed with an exception.

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30

* Try:
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s

    at makeError (C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp\node_modules\@react-native-community\cli-platform-android\node_modules\execa\index.js:174:9)
    at C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp\node_modules\@react-native-community\cli-platform-android\node_modules\execa\index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:82:7)
    at async Command.handleAction (C:\Users\Szabolcs\Desktop\Szaby\Projects\TSISApp2\tsisappp\node_modules\@react-native-community\cli\build\index.js:108:9)
info Run CLI with --verbose flag for more details.
```

### Device

Huawei Honor 9X Android

### VisionCamera Version

"react-native-vision-camera": "^2.15.4"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

javier-sauma-rocketlab, 2023-03-14:
same over here

germanolira, 2023-03-15:
If you are using Expo, it's recommended that you use the following command:
`expo install react-native-vision-camera`

Expo docs:
**"We recommend always using npx expo install instead of npm install or yarn add directly because it allows Expo CLI to pick a compatible version of a library when possible and also warn you about known incompatibilities."**

tsanyqudsi, 2023-03-28:
to add from @javier-sauma-rocketlab 's answer.

If you're using expo, install expo-build-properties ( https://docs.expo.dev/versions/latest/sdk/build-properties/ ) to change kotlin version

```
{
  "expo": {
    "plugins": [
      [
        "expo-build-properties",
        {
          "android": {
           ...,
           "kotlinVersion": "1.7.0"
          }
        }
      ]
    ]
  }
}
```

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

andriyfm, 2023-05-26:
Same issue

Szabolcs11, 2023-07-04:
I just added kotlinVersion = "1.7.0" to android/build.gradle and now it's working.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy