margelo/react-native-vision-camera #1524: 🐛 The Android extensions ('kotlin-android-extensions') compiler plugin is no longer supported.

Flonk, 2023-03-15:
### What were you trying to do?

Build an expo app.

I'm not exactly privvy to app development, here are some dependencies and their versions:

```
"dependencies": {
    "detox": "20.5.0",
    "expo": "~47.0.12",
    "react-native": "0.70.5",
    "react-native-vision-camera": "2.15.4",
  }
```

Kotlin version 1.8.0
Gradle version 8.0 (I think)

I'm using detox to build the app, not sure if this matters.

### Reproduceable Code

```tsx
-
```

### What happened instead?

```
λ> detox build --configuration android.emu.release

cd android && ./gradlew assembleRelease assembleAndroidTest -DtestBuildType=release

> Task :react-native-vision-camera:compileReleaseKotlin FAILED
e: The Android extensions ('kotlin-android-extensions') compiler plugin is no longer supported. Please use kotlin parcelize and view binding. More information: https://goo.gle/kotlin-android-extensions-deprecation
```

### Device

Ubuntu 22.04

### VisionCamera Version

2.15.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Flonk, 2023-03-15:
Solution: remove the line `apply plugin: 'kotlin-android-extensions'` from react-native-vision-camera's `build.gradle` file.

closed 2023-03-15 by Flonk

fontesrp, 2023-03-17:
We should probably open a PR to fix this. As of now, the only reference to the error is this closed issue, not the easiest to find.

ArtanAhmeti, 2023-07-13:
> Solution: remove the line `apply plugin: 'kotlin-android-extensions'` from react-native-vision-camera's `build.gradle` file.

Thank you! This helped after upgrading Kotlin to 1.8.0 from 1.6.0.