margelo/react-native-vision-camera #1525: When open camera surface in RedMi 10T 5G app crashes🐛

iasreact1, 2023-03-17:
### What were you trying to do?

@ChristophGr  When open camera for recording video in RedMi 10T 5G device it crashes generating a huge bug report 
The Bug report is given below & my simple coding is below
` const devices = useCameraDevices();
 const [isFont, setIsFont] = useState(true);
let deviceId = isFont == true ? devices.back : devices.front
//checking for camera and microphone permission
if (device == null) {
        return <ActivityIndicator size={20} color={'red'} />;
      }
return(
 <View style={{ height: height}}>
                <Camera
                    ref={camera}
                    style={StyleSheet.absoluteFill}
                    isActive={isRecording}
                    video={true}
                    audio={true}
                    device={deviceId}
                />
            </View>
);
`

### What happened instead?

java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.ResourceCreationException: androidx.camera.video.internal.encoder.InvalidConfigException: Encoder cannot created: null
	at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:406)
	at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:416)
	at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:482)
	at androidx.camera.video.VideoCapture.lambda$createPipeline$0$androidx-camera-video-VideoCapture(VideoCapture.java:433)
	at androidx.camera.video.VideoCapture$$ExternalSyntheticLambda0.onError(Unknown Source:10)
	at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$14(Camera2CameraImpl.java:1175)
	at androidx.camera.camera2.internal.Camera2CameraImpl$$ExternalSyntheticLambda7.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:210)
	at android.os.Looper.loop(Looper.java:299)
	at android.app.ActivityThread.main(ActivityThread.java:8306)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:576)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1073)

### Device

RedMI 10T 5G

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

valn1, 2023-04-03:
I am also having this issue in my app.
some specific devices throw this error when opening the camera, all of them in android 12.
POCO camellian, Redmi lancelot, Redmi miel.

this is the logcat for the issue:

```
Exception java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.ResourceCreationException: androidx.camera.video.internal.encoder.InvalidConfigException: Encoder cannot created: null
  at androidx.camera.video.Recorder.onSurfaceRequested (Recorder.java:406)
  at androidx.camera.video.VideoCapture.createPipeline (VideoCapture.java:416)
  at androidx.camera.video.VideoCapture.resetPipeline (VideoCapture.java:482)
  at androidx.camera.video.VideoCapture.lambda$createPipeline$0$androidx-camera-video-VideoCapture (VideoCapture.java:433)
  at androidx.camera.video.VideoCapture$$ExternalSyntheticLambda0.onError
  at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$14 (Camera2CameraImpl.java:1175)
  at androidx.camera.camera2.internal.Camera2CameraImpl$$ExternalSyntheticLambda13.run
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:211)
  at android.os.Looper.loop (Looper.java:300)
  at android.app.ActivityThread.main (ActivityThread.java:8282)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:576)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1074)
```

i am using react-native-vision-camera v2.15.4

valn1, 2023-04-17:
```
    implementation "androidx.camera:camera-core:1.2.0-beta02"
    implementation "androidx.camera:camera-camera2:1.2.0-beta02"
    implementation "androidx.camera:camera-lifecycle:1.2.0-beta02"
    implementation "androidx.camera:camera-video:1.2.0-beta02"
    implementation "androidx.camera:camera-view:1.2.0-beta02"
    implementation "androidx.camera:camera-extensions:1.2.0-beta02"
```
place this inside your android/app/build.gradle inside dependencies prop.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy