margelo/react-native-vision-camera #1530: How can I take close-up photos that are still clear on a phone that only has "ultra-wide-angle-camera"❓

vuhongphucvn, 2023-03-20:
### Question

Hi everyone, 
I'm doing a function about take a photo, however when i take a photo of the ID card at close range by back camera, it's very blurry and can't be seen clearly. I'm using Samsung Galaxy S22 device, i traced the getAvailableCameraDevices() method and it just returns the result as "ultra-wide-angle-camera" for back camera. When i try to shoot with phones with "wide-angle-camera" everything is clear.

How can I take close-up photos that are still clear on a phone that only has "ultra-wide-angle-camera"?
Thank!

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jianjiax, 2023-07-27:
same issues

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy