margelo/react-native-vision-camera #1531: 🐛 all devices are undefined on `Smartisan` phone.

chj-damon, 2023-03-20:
### What were you trying to do?

I'm trying to use vision-camera on my Smartisan cellphone. 
```ts
const devices = useCameraDevices();
// or
const devices = await Camera.getAvailableCameraDevices();
```
all devices are undefined:
```
{"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}
```

it throws an error here:
```
unknown/unknown: [unknown/unknown] cameraConfig.getOutputSizes(it) must not be null
```

I think this may be related to #281 but I have no idea.

### What happened instead?

app cannot work because this error

### Device

smartisan OS OE106 (based on android 8.1.0)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vadymsushkovdev, 2023-05-26:
Have the same issue on Poco phone. I'm using `  const devices = useCameraDevices();
  const cameraDevice = devices.back;` but getting `{"cameraDevice": undefined, "devices": {"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}}`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

GolliAch, 2023-12-28:
@mrousavy Have been having the same issue on iPhone using V3. V2 works fine, we are trying to migrate to V3

mrousavy, 2023-12-29:
@GolliAch  The original issue was reported for V2, are you 100% sure that V2 worked and V3 doesn't work? Please share your code and the console.log output you get.