margelo/react-native-vision-camera #1532: 🐛 Crashing on fast start / stopRecording calls - Unexpectedly found nil while unwrapping an Optional value

vntech-engineering69, 2023-03-21:
### What were you trying to do?

When I try to create a couple of short video shots (<1s) with a more rapid "tapping" approach, and crash. even if I wait a bit more before I hit the button.

### Reproduceable Code

```tsx
Just called startRecord / stopRecording quite frequently. Will happen at some point (sometimes after the 5th try, sometimes after the 15th hit).
```

### What happened instead?

VisionCamera/CameraViewManager.swift:179: Fatal error: Unexpectedly found nil while unwrapping an Optional value
[image]
[image]

### Relevant log output

```shell
2023-03-21 14:54:04.773368+0530 DEV-contentCreator[6785:1657583] [native] (ADVICE) View #179 of type RCTView has a shadow set but cannot calculate shadow efficiently. Consider setting a background color to fix this, or apply the shadow to a more specific component.
2023-03-21 14:54:04.774549+0530 DEV-contentCreator[6785:1657583] [native] (ADVICE) View #163 of type RCTView has a shadow set but cannot calculate shadow efficiently. Consider setting a background color to fix this, or apply the shadow to a more specific component.
2023-03-21 14:54:04.775130+0530 DEV-contentCreator[6785:1657583] [native] (ADVICE) View #127 of type RCTView has a shadow set but cannot calculate shadow efficiently. Consider setting a background color to fix this, or apply the shadow to a more specific component.
2023-03-21 14:54:08.397394+0530 DEV-contentCreator[6785:1657583] [native] VisionCamera.didSetProps(_:): Updating 19 prop(s)...
2023-03-21 14:54:08.397512+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.configureAudioSession(): Configuring Audio Session...
2023-03-21 14:54:08.399139+0530 DEV-contentCreator[6785:1657583] [native] [GESTURE HANDLER] Initialize gesture handler for view <RCTRootContentView: 0x10810a0b0; reactTag: 11; frame = (0 0; 390 844); gestureRecognizers = <NSArray: 0x2839b4ed0>; layer = <CALayer: 0x2837843a0>>
2023-03-21 14:54:08.399406+0530 DEV-contentCreator[6785:1657583] [native] VisionCamera.didSetProps(_:): Updating 1 prop(s)...
2023-03-21 14:54:08.399480+0530 DEV-contentCreator[6785:1657917] [native] VisionCamera.configureAudioSession(): Configuring Audio Session...
2023-03-21 14:54:08.399552+0530 DEV-contentCreator[6785:1657917] [native] VisionCamera.configureAudioSession(): Adding Audio input...
2023-03-21 14:54:08.406115+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2023-03-21 14:54:08.406135+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
2023-03-21 14:54:08.406143+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2023-03-21 14:54:08.409549+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2023-03-21 14:54:08.410158+0530 DEV-contentCreator[6785:1657917] [native] VisionCamera.configureAudioSession(): Adding Audio Data output...
2023-03-21 14:54:08.410499+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2023-03-21 14:54:08.410790+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2023-03-21 14:54:08.411237+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureFormat(): Configuring Format...
2023-03-21 14:54:08.412455+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureFormat(): Format successfully configured!
2023-03-21 14:54:08.412463+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureDevice(): Configuring Device...
2023-03-21 14:54:08.412519+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.configureDevice(): Device successfully configured!
2023-03-21 14:54:08.412906+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.didSetProps(_:): Starting Session...
2023-03-21 14:54:08.635033+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.didSetProps(_:): Started Session!
2023-03-21 14:54:09.280789+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.startRecording(options:callback:): Starting Video recording...
2023-03-21 14:54:09.287071+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.startRecording(options:callback:): File path: /private/var/mobile/Containers/Data/Application/79740389-95FE-4820-968A-3BA85BD2F621/tmp/ReactNative/28E9F06C-27C7-46C5-A29D-35740B24558D.mov
2023-03-21 14:54:09.299448+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.initializeVideoWriter(withSettings:pixelFormat:): Initialized Video AssetWriter.
2023-03-21 14:54:09.299510+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.activateAudioSession(): Activating Audio Session...
2023-03-21 14:54:09.301256+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.updateCategory(_:options:): Changing AVAudioSession category from AVAudioSessionCategorySoloAmbient -> AVAudioSessionCategoryPlayAndRecord
2023-03-21 14:54:09.637078+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.initializeAudioWriter(withSettings:): Initialized Audio AssetWriter.
2023-03-21 14:54:09.637094+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.startAssetWriter(): Starting Asset Writer(s)...
2023-03-21 14:54:09.868487+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.appendBuffer(_:type:timestamp:): Started RecordingSession at 325881.09776575 seconds.
2023-03-21 14:54:11.202122+0530 DEV-contentCreator[6785:1657921] [native] VisionCamera.finish(): Finishing Recording with AssetWriter status "writing"...
2023-03-21 14:54:11.249455+0530 DEV-contentCreator[6785:1657916] [native] VisionCamera.startRecording(options:callback:): RecordingSession finished with status completed.
2023-03-21 14:54:11.249661+0530 DEV-contentCreator[6785:1657909] [native] VisionCamera.deactivateAudioSession(): Deactivating Audio Session...
2023-03-21 14:54:11.305562+0530 DEV-contentCreator[6785:1657583] [native] VisionCamera.didSetProps(_:): Updating 1 prop(s)...
2023-03-21 14:54:11.310591+0530 DEV-contentCreator[6785:1657914] [native] VisionCamera.didSetProps(_:): Stopping Session...
2023-03-21 14:54:11.457834+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | ┌──────────────────────────────────────────────────────────────────────────────────────────┐
2023-03-21 14:54:11.457913+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ The VideoEditor SDK is a commercial product. To use it as such and get access to its     │
2023-03-21 14:54:11.458002+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ white label version - without the IMG.LY logo and watermark rendered - you need to       │
2023-03-21 14:54:11.458042+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ unlock the SDK with a license file. You can purchase a license at https://img.ly/pricing │
2023-03-21 14:54:11.458082+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | └──────────────────────────────────────────────────────────────────────────────────────────┘
2023-03-21 14:54:11.458120+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | ┌──────────────────────────────────────────────────────────────────────────────────────────┐
2023-03-21 14:54:11.458156+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ Because you did not specify a license file yet, you are in the Free Trial mode. The SDK  │
2023-03-21 14:54:11.458192+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ will now display a watermark image on top of any videos you display or export with it.   │
2023-03-21 14:54:11.458226+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ For instructions on how to unlock the SDK, please visit:                                 │
2023-03-21 14:54:11.458257+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | │ https://img.ly/docs/vesdk/ios/introduction/getting_started/#license-file.                │
2023-03-21 14:54:11.458322+0530 DEV-contentCreator[6785:1658053] [ImglyKit] ERR | └──────────────────────────────────────────────────────────────────────────────────────────┘
2023-03-21 14:54:11.471409+0530 DEV-contentCreator[6785:1657583] Metal API Validation Enabled
2023-03-21 14:54:12.157399+0530 DEV-contentCreator[6785:1657914] [native] VisionCamera.didSetProps(_:): Stopped Session!
[... cut ...]
### Device

iphone 12

### VisionCamera Version

^2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vntech-engineering69, 2023-03-21:
I have made some changes in the swift just for testing purposes. 
`// pragma MARK: React Functions

  @objc
  final func startRecording(_ node: NSNumber, options: NSDictionary, onRecordCallback: @escaping RCTResponseSenderBlock) {
    let component = getCameraView(withTag: node)
    component?.startRecording(options: options, callback: onRecordCallback)
  }

  @objc
  final func pauseRecording(_ node: NSNumber, resolve: @escaping RCTPromiseResolveBlock, reject: @escaping RCTPromiseRejectBlock) {
    let component = getCameraView(withTag: node)
    component?.pauseRecording(promise: Promise(resolver: resolve, rejecter: reject))
  }

  @objc
  final func resumeRecording(_ node: NSNumber, resolve: @escaping RCTPromiseResolveBlock, reject: @escaping RCTPromiseRejectBlock) {
    let component = getCameraView(withTag: node)
    component?.resumeRecording(promise: Promise(resolver: resolve, rejecter: reject))
  }

  @objc
  final func stopRecording(_ node: NSNumber, resolve: @escaping RCTPromiseResolveBlock, reject: @escaping RCTPromiseRejectBlock) {
    let component = getCameraView(withTag: node)
    component?.stopRecording(promise: Promise(resolver: resolve, rejecter: reject))
  }

  @objc
  final func takePhoto(_ node: NSNumber, options: NSDictionary, resolve: @escaping RCTPromiseResolveBlock, reject: @escaping RCTPromiseRejectBlock) {
    let component = getCameraView(withTag: node)
    component?.takePhoto(options: options, promise: Promise(resolver: resolve, rejecter: reject))
  }

  @objc
  final func focus(_ node: NSNumber, point: NSDictionary, resolve: @escaping RCTPromiseResolveBlock, reject: @escaping RCTPromiseRejectBlock) {
    let promise = Promise(resolver: resolve, rejecter: reject)
    guard let x = point["x"] as? NSNumber, let y = point["y"] as? NSNumber else {
      promise.reject(error: .parameter(.invalid(unionName: "point", receivedValue: point.description)))
      return
    }
    let component = getCameraView(withTag: node)
    component?.focus(point: CGPoint(x: x.doubleValue, y: y.doubleValue), promise: promise)
  }
  
      private func getCameraView(withTag tag: NSNumber) -> CameraView? {
      return bridge.uiManager.view(forReactTag: tag) as? CameraView
    }

  `
  I am not that good at swift. 
  
  I am currently testing this fix as there is no way to know when it crashes.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy