margelo/react-native-vision-camera #1538: 🐛 cannot find symbol new CameraPackage() [com.mrousavy.camera.example.CameraPackage]

AliMSbaee, 2023-03-22:
### What were you trying to do?

Just updated to RN@0.71.4 and React-Native-Reanimated@3.0.2 and got this 2 errors. I've also tried to manually link `react-native-vision-camera` by following the example on `v3` branch but with no luck. Am I missing something?

### What happened instead?

Build failed with an exception.

### Relevant log output

```shell
/Users/ali/Documents/flipnpik-mobile/android/app/build/generated/rncli/src/main/java/com/facebook/react/PackageList.java:85: error: cannot find symbol
import com.mrousavy.camera.example.CameraPackage;
                                  ^
  symbol:   class CameraPackage
  location: package com.mrousavy.camera.example
/Users/ali/Documents/flipnpik-mobile/android/app/build/generated/rncli/src/main/java/com/facebook/react/PackageList.java:170: error: cannot find symbol
      new CameraPackage(),
          ^
  symbol:   class CameraPackage
  location: class PackageList
```

### Device

Android Emulator (API 29)

### VisionCamera Version

3.0.0.rc-2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

AliMSbaee, 2023-03-27:
> @AliMSbaee how were you able to fix this?

install from `v3` branch
`npm install "https://github.com/mrousavy/react-native-vision-camera.git#v3" --save`

hkeithk, 2023-03-27:
did you have any issues installing worklets? 

[image]

[image]

AliMSbaee, 2023-03-28:
@hkeithk  
ty to install worklets from here `https://github.com/chrfalch/react-native-worklets#d62d76c`

tientruongvan1995, 2023-04-07:
> > @AliMSbaee how were you able to fix this?
> 
> install from `v3` branch `npm install "https://github.com/mrousavy/react-native-vision-camera.git#v3" --save`

I received error: 
`Cannot find module 'react-native-vision-camera' or its corresponding type declarations`

lucienbl, 2023-05-30:
Hello, 

I had the same issue using `3.0.0-rc.2` and it looks like there is a small typo in the `AndroidManifest.xml` file, here is my patch:
```diff
diff --git a/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml b/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml
index 2906763..f5a395b 100644
--- a/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml
+++ b/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml
@@ -1,4 +1,4 @@
 <manifest xmlns:android="http://schemas.android.com/apk/res/android"
-          package="com.mrousavy.camera.example">
+          package="com.mrousavy.camera">
 
 </manifest>
 ```
 I can successfully build on Android after fixing that. Hope this can help some of you :)
 
 Best regards.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy