margelo/react-native-vision-camera #1539: 🐛 2.15.3+ has a 90 rotation on landscape iOS

iBotPeaches, 2023-03-23:
### What were you trying to do?

Take a photo in landscape, which is now rotated (90) and wrong after updating from 2.15.2 to 2.15.4.

Reverted this from 2.15.4 (to keep the RN71 fix) and all was resolved - https://github.com/mrousavy/react-native-vision-camera/commit/bf71901968fcdb7fbb5864c435ff239df23c81a5

### Reproduceable Code

```tsx
<Camera
          ref={scannerRef}
          photo={photo}
          device={device}
          isActive={isFocused}
        />
```

### What happened instead?

The image displayed in preview is 90 degrees wrong, despite not specifying a rotation.

### Relevant log output

```shell
n/a
```

### Device

iPad 6 Gen

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iBotPeaches, 2023-03-23:
@mrousavy - Trying to help out with a patch, but not following why this changed to be locked to portrait in this commit - https://github.com/mrousavy/react-native-vision-camera/commit/bf71901968fcdb7fbb5864c435ff239df23c81a5#diff-bff70569c755a3c0200dca5fff5c86d56017e222b6e7d2d1d348b6992d623cbeR15

Any ideas?

nn1900, 2023-04-04:
Also experienced this issue on iPad, have to downgrade to 2.14.1 temporarily, waiting for a fix for this.

hritiksingh73, 2023-07-04:
@mrousavy any update ,fix or workaround on this will be highly appriaciated

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
JFYI; Camera sensors always have a 90deg orientation

PR #1544 (2023-09-30): fix: updated orientation calculations
## What

This PR should fix the issues in video orientation (preview and recording) being "fixed" to portrait orientation regardless of device orientation in iOS.

Adapted from https://stackoverflow.com/a/36575423

## Changes

* Changed the CameraView orientation extension so that it takes the UIDevice's reported orientation if the userOrientation is not provided
* On rotation change it will update the preview layer orientation and update the preview layer bounds accordingly

## Tested on

* iPhone 14 Pro
* iPad Mini

## Related issues

* Fixes #1539 
* Fi [... cut]
closed 2023-09-30 by mrousavy