margelo/react-native-vision-camera #1540: ❓Should you be able to set a custom video width / height?

Sid-Turner-Ellis, 2023-03-24:
### Question

Hey,

Im trying to reduce the video file size by reducing the quality of the recorded video. My device only returns 1 format when logging `useCameraDevices().devices.back.formats` even though my camera app supports qualities up to 4k.

Im trying to set the format myself using something like:
```
const format = getFormat...

const scale = 0.5
format.videoWidth = 1280 * scale
format.videoHeight = 720 * scale
```

It works when showing the camera preview and the correct quality is showing however when I call `startRecording()` I get an inactive source error. I find this strange because[ I can see on this line](https://github.com/mrousavy/react-native-vision-camera/blob/ba099e4044714a2178eff9c6ef148420124cd416/android/src/main/java/com/mrousavy/camera/CameraView.kt#L424) that the exact size aren't even being used and it should fallback yet even doing something like this causes the error:

```
format.videoWidth = 1280 - 1
format.videoHeight = 720 - 1
```

Anyone know whats going on here?

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

garmoshka-mo, 2023-04-07:
Photo & video sizes not configurable neither with presets nor with format props. (With format you can actually change size - but it will be random, not even close to width/height of specified format).

I had to hardcode desired resolution with patch: https://github.com/mrousavy/react-native-vision-camera/issues/1548#issuecomment-1500208455

Sid-Turner-Ellis, 2023-04-07:
Thanks @garmoshka-mo

TexxUkr, 2023-05-26:
@garmoshka-mo 
unfortunately the patch proposed does not work for all devices I tried
it caused the following error:
"Failed to configure session: Unable to find supported quality by QualitySelector update() threw: [device/invalid-device] The given Camera device could not be found for use-case binding!
invokeOnError(...):"
As far as I understand there is no other workaround than convert saved recording file by ffmpeg-kit...

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
Actually no you shouldn't edit format.

closed 2023-09-30 by mrousavy