margelo/react-native-vision-camera #1542: 🐛 Android 12 Frame Processor Plugin not getting image from the camera

amf-paulinho, 2023-03-26:
### What were you trying to do?

Hello !
I created a frame processor to read ISBN barcode using ZXING library. It is working and stable (over 6 months) on Android 11 and below. now On Android 12 does not work. Only phone I tested was Moto G Stylus. To debug the problem I decided to save on the disk the image received by the frame processor plugin and I found the problem. 

### Reproduceable Code

```tsx
package com.bookscanner.upcframeprocessor;

import android.content.res.Configuration;
import android.graphics.Bitmap;
import android.graphics.BitmapFactory;
import android.graphics.ImageFormat;
import android.graphics.Rect;
import android.graphics.YuvImage;
import android.util.Log;

import androidx.annotation.NonNull;
import androidx.camera.core.ImageProxy;

import com.google.zxing.BinaryBitmap;
import com.google.zxing.DecodeHintType;
import com.google.zxing.FormatException;
import com.google.zxing.NotFoundException;
import com.google.zxing.PlanarYUVLuminanceSource;
import com.google.zxing.Result;
import com.google.zxing.ResultMetadataType;
import com.google.zxing.common.HybridBinarizer;
import com.google.zxing.oned.MultiFormatUPCEANReader;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.io.ByteArrayOutputStream;
import java.io.File;
import java.io.FileOutputStream;
import java.io.IOException;
import java.nio.ByteBuffer;
import java.text.SimpleDateFormat;
import java.util.Date;
import java.util.HashMap;

class ImageData {
  public int width;
  public int height;
  public byte[] data;
}

public class UpcFrameProcessorPlugin extends FrameProcessorPlugin {
  Configuration _config;

  private String saveToFile(ImageProxy image) {

    try {

      SimpleDateFormat dateFormat = new SimpleDateFormat("yyyyMMdd_HH_mm_ss");
      String currentTimeStamp = dateFormat.format(new Date());

      // Get path of image to load
      String path = "/storage/emulated/0/picture_" + currentTimeStamp + ".png";

      ImageProxy.PlaneProxy[] planes = image.getPlanes();

      ByteBuffer yBuffer = planes[0].getBuffer();
      ByteBuffer uBuffer = planes[1].getBuffer();
      ByteBuffer vBuffer = planes[2].getBuffer();

      int ySize = yBuffer.remaining();
      int uSize = uBuffer.remaining();
      int vSize = vBuffer.remaining();

      byte[] nv21 = new byte[ySize + uSize + vSize];
      yBuffer.get(nv21, 0, ySize);
      vBuffer.get(nv21, ySize, vSize);
      uBuffer.get(nv21, ySize + vSize, uSize);

      YuvImage yuvImage = new YuvImage(nv21, ImageFormat.NV21, image.getWidth(), image.getHeight(), null);
      ByteArrayOutputStream byteArrayOut = new ByteArrayOutputStream();
      yuvImage.compressToJpeg(new Rect(0, 0, yuvImage.getWidth(), yuvImage.getHeight()), 75, byteArrayOut);

      byte[] imageBytes = byteArrayOut.toByteArray();
      Bitmap bm = BitmapFactory.decodeByteArray(imageBytes, 0, imageBytes.length);

      File file = new File(path);
      FileOutputStream out = new FileOutputStream(file);
      bm.compress(Bitmap.CompressFormat.JPEG, 90, out);
      out.flush();
      out.close();
      System.out.println("Correctly saved");

      return "image";

    } catch (Exception e) {
      System.out.println("CRASHED");
      e.printStackTrace();
      return e.getMessage();
    }

  }

  private byte[] imageProxyToByteArray(ImageProxy image) {
    ImageProxy.PlaneProxy[] planes = image.getPlanes();
    ByteBuffer buff = planes[0].getBuffer();
    buff.rewind();
    byte[] arr = new byte[buff.remaining()];
    buff.get(arr);

    return arr;
  }

  /*
   * The Library consider any image as landscape even if the snap was taken with
   * the phone in portrait
   * Because this is the default behavior for phone cameras.
   * Knowing that, if the phone Orientation is Portrait its necessary to rotate
   * the image 90 degrees
   */
  private ImageData rotateImageIfNecessary(ImageProxy image) {
    byte[] imageBytes = imageProxyToByteArray(image);
    byte[] rotatedData = new byte[imageBytes.length];
    int imageWidth = image.getWidth();
    int imageHeight = image.getHeight();

    ImageData imageData = new ImageData();

    if (_config.orientation == Configuration.ORIENTATION_PORTRAIT && image.getImageInfo().getRotationDegrees() == 90) {
      for (int y = 0; y <= imageHeight - 1; y++) {
        for (int x = 0; x <= imageWidth - 1; x++) {
          rotatedData[x * imageHeight + imageHeight - y - 1] = imageBytes[x + y * imageWidth];
        }
      }

      imageData.width = imageHeight;
      imageData.height = imageWidth;
      imageData.data = rotatedData;

    } else {
      imageData.width = imageWidth;
      imageData.height = imageHeight;
      imageData.data = imageBytes;
    }

    return imageData;
  }

  private PlanarYUVLuminanceSource prepareYUVSource(ImageProxy image) {

    // Obtain Frame configuuratio (Used for cropping the image)
    // ZebraConstantsFactory constantsFactory = new ZebraConstantsFactory();
    // IFRAME_PROCESSOR_CONSTANTS _CONSTANTS =
    // constantsFactory.getConstants(_config.orientation);

    double FRAME_TOP_SPACING_PERCENT = 0; // 25.0;
    double FRAME_BOTTOM_SPACING_PERCENT = 0; // 35.0;
    double FRAME_HORIZONTAL_SPACING_PERCENT = 0;// 5.0;

    // get Image Array of Bytes and Rotate Image if necessary
    ImageData transformedImage = rotateImageIfNecessary(image);

    // Preparing crop variables
    double finalRectWidth = transformedImage.width * (1 - FRAME_HORIZONTAL_SPACING_PERCENT / 100);
    double finalRectHeight = transformedImage.height
        * (1 - (FRAME_TOP_SPACING_PERCENT + FRAME_BOTTOM_SPACING_PERCENT) / 100);
    double top = transformedImage.height * FRAME_TOP_SPACING_PERCENT / 100;
    double left = transformedImage.width * FRAME_HORIZONTAL_SPACING_PERCENT / 2 / 100;

    // Crop the image
    return new PlanarYUVLuminanceSource(
        transformedImage.data,
        transformedImage.width,
        transformedImage.height,
        (int) left,
        (int) top,
        (int) finalRectWidth,
        (int) finalRectHeight,
        false);
  }

  @Override
  public Object callback(ImageProxy image, Object[] params) {

    String x = this.saveToFile(image);

    HashMap<DecodeHintType, int[]> map = new HashMap<>();
    map.put(DecodeHintType.ALLOWED_EAN_EXTENSIONS, new int[] { 5 });

    MultiFormatUPCEANReader reader = new MultiFormatUPCEANReader(map);
    PlanarYUVLuminanceSource source = this.prepareYUVSource(image);
    BinaryBitmap binaryBitmap = new BinaryBitmap(new HybridBinarizer(source));
    try {
      // Whenever reader fails to detect a QR code in image
      // it throws NotFoundException

      Result result = reader.decode(binaryBitmap);
      String upc = result.getText();
      Object ean = result.getResultMetadata().get(ResultMetadataType.UPC_EAN_EXTENSION);

      if (ean != null) {
        return upc.substring(0, upc.length() - 1) + ean;
      }

      return "";
    } catch (NotFoundException | FormatException e) {

      e.printStackTrace();

      return "";
    } finally {
      image.close();
    }

    // return x;
  }

  public UpcFrameProcessorPlugin(android.content.Context context) {
    super("scanUpc");
    _config = context.getResources().getConfiguration();

  }
}
```

### What happened instead?

No error, no crash. the problem is I am instead getting the barcode picture to process. I am getting a stripped distorted image with random color, like a old Tube TV statics . Obviously ZXing can't find a barcode there because there are none. this image is from the internet but it describe very well what  I am getting inside ImageProxy: https://i.stack.imgur.com/uk3eG.jpg

### Device

Moto G Stylus

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

amf-paulinho, 2023-03-26:
This post was a big clue for me: https://gist.github.com/Ahwar/b6797f81671b2f8fb3f7cc5de3c9a5dc

zzz08900, 2023-04-01:
Definitely has nothing to do with vision camera. My guess is google did something wired on Android 12 and screwed your code.
I'm using android.media.Image with no problem right now. Like this:
`Image mediaImage = frame.getImage();`

amf-paulinho, 2023-04-07:
> Definitely has nothing to do with vision camera. My guess is google did something wired on Android 12 and screwed your code. I'm using android.media.Image with no problem right now. Like this: `Image mediaImage = frame.getImage();` The resulting android.media.Image works fine with MLKit and other library that takes android.media.Image as input.

You are correct. I did some research on-line  and found out more people with same problem however nothing to do with Camera vision. I had to change the way I work working with the Image Proxy

closed 2023-04-07 by amf-paulinho