margelo/react-native-vision-camera #1547: 🐛 RN 0.71.4 run-android ==> Task :react-native-vision-camera:compileDebugKotlin FAILED

thuantvDr4, 2023-03-30:
### What were you trying to do?

I am trying to install this library
"react-native-vision-camera": "2.15.4",

    "react": "18.2.0",
    "react-native": "0.71.4",

### Reproduceable Code

```tsx
''
```

### What happened instead?

> Task :react-native-vision-camera:compileDebugKotlin FAILED

### Relevant log output

```shell
FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
java.lang.StackOverflowError (no error message)

> Task :react-native-vision-camera:compileDebugKotlin FAILED

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 44s
```

### Device

android

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Lonoyaa, 2023-03-31:
Hi, have you try running assembleRelease with --info for more detail
I'm just happened to work with the same version of dependencies as you but mine's fine. Did you try adding kotlinVersion and kotlin-gradle-plugin to android/build.gradle? if so, what version is it? mine is 1.6.20

roberto497, 2023-04-01:
Same problem!

zzz08900, 2023-04-01:
Probably related https://github.com/gradle/gradle/issues/21116#issuecomment-1228005104

Ho3ein-Boka, 2023-04-04:
same problem

thuantvDr4, 2023-04-06:
thanks all !

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

diamont1001, 2023-04-23:
> changing kotlin version worked for me
> 
> ```groovy-gradle
> /*
> /android/build.gradle
> */
> buildscript {
>     ext {
>     // ...
>         kotlinVersion = "1.6.20"
>     }
>     // ...
> }
> ```

it works, thank you!

adscud, 2023-05-05:
Hello, any workaround without downgrading the kotlin version ?

kigh143, 2023-07-19:
i am getting the same issue with version 0.65 RN

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

xts-bit, 2023-09-30:
@mrousavy This issue is still coming on V3.2 latest one

ISnowFoxI, 2023-10-16:
Removing `kotlinVersion` from `build.gradle` fixed the issue for me. I added it because of another library if I remember correctly. You could also try the `build.gradle` code from the example.

https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/android/build.gradle#L5-L10

tatiana-lonestone, 2023-10-30:
Hello !

@mrousavy This issue is still coming on V3.6.4 on android

* React Native => 0.71.14
* Reanimated => 3.5.4

````
buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 28
        compileSdkVersion = 33
        targetSdkVersion = 33
        supportLibVersion = "28.0.0"
        kotlinVersion = "1.6.20"
        ndkVersion = "23.1.7779620"
````
[image]

tatiana-lonestone, 2023-10-31:
Hi !
I changed `kotlinVersion` for 1.7.20 and this is now working !

Current working for react-native-vision-camera@3.6.4 and react-native@0.71.14 :

````
buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 28
        compileSdkVersion = 33
        targetSdkVersion = 33
        supportLibVersion = "28.0.0"
        kotlinVersion = "1.7.20"
        ndkVersion = "23.1.7779620"
````

ShuraRB, 2023-11-09:
Changing the kotlinVersion to 1.7.20 also worked for me!!1

kelrvins, 2023-12-10:
> Hi ! I changed `kotlinVersion` for 1.7.20 and this is now working !
>
> Current working for react-native-vision-camera@3.6.4 and react-native@0.71.14 :
>
> ```
> buildscript {
>     ext { [...]

Thank you very much, I spent three days on this issue

robertyulisman, 2023-12-12:
> Hi ! I changed `kotlinVersion` for 1.7.20 and this is now working !
>
> Current working for react-native-vision-camera@3.6.4 and react-native@0.71.14 :
>
> ```
> buildscript {
>     ext { [...]

its work fine for me, thanks so much

zahid502, 2024-06-10:
Hi, added some code in node_modules of react-native-vision-camera and it's worked for me.
**react-native 0.74.1**

**path**: node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt

`import com.facebook.react.common.annotations.FrameworkAPI//<--import here...
@OptIn(FrameworkAPI::class)//<--insert here`

[image]

mbalazy, 2024-08-21:
> Hi, added some code in node_modules of react-native-vision-camera and it's worked for me. **react-native 0.74.1**
> 
> **path**: node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
> 
> `import com.facebook.react.common.annotations.FrameworkAPI//<--import here... @OptIn(FrameworkAPI::class)//<--insert here`
> 
> <img alt="Screenshot 2024-06-10 at 10 19 30 PM" width="1189" src="https://private-user-images.githubusercontent.com/115239786/338282720-daa51432-788d-4591-b696-6af7f7ad66e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyMzc5NDUsIm5iZiI6MTcyNDIzNzY0NSwicGF0aCI6Ii8xMTUyMzk3ODYvMzM4MjgyNzIwLWRhYTUxNDMyLTc4OGQtNDU5MS1iNjk2LTZhZjdmN2FkNjZlMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyMVQxMDU0MDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mOTI4ZTRmNGMzMTg1NzU3MzNmZjYwZGI2Y2FiYWI5NDVhMjAyYjQxMGM4NTllYjRiMDM4MjVhODYyMGY5NjAwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.M3tWXKNY9WnfUJGxczf4GO8i0zMHPDSu_YUQs69J5Nk">

thanks!

I created a diff for that:

```diff
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt
index 81bd1af..dbe53be 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt
@@ -12,7 +12,9 @@ import com.facebook.react.uimanager.UIManagerHelper
 import com.mrousavy.camera.CameraView
 import com.mrousavy.camera.core.ViewNotFoundError
 import java.lang.ref.WeakReference
+import com.facebook.react.common.annotations.FrameworkAPI
 
+@OptIn(FrameworkAPI::class)
 @Suppress("KotlinJniMissingFunction") // we use fbjni.
 class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
   companion object {

```

banezr94, 2025-02-01:
Same issue for me, using the latest version of RN.

davidburson, 2025-02-01:
@banezr94 I ran this script:
`"clean-after-rn-upgrade": "node -e \"fs.rmSync('./node_modules', { recursive: true, force: true })\" && pnpm store prune && pnpm install && cd android && gradlew clean && cd .. && react-native start --reset-cache",`

Then the patch mentioned in these got me past this issue:
https://github.com/mrousavy/react-native-vision-camera/issues/3389
https://github.com/mrousavy/react-native-vision-camera/pull/3382/files
https://github.com/mrousavy/react-native-vision-camera/issues/3380

But now I have this issue: https://github.com/mrousavy/react-native-vision-camera/issues/2862

So I may just wait until RN 0.78.0 and the next version of RNVC:  https://github.com/mrousavy/react-native-vision-camera/issues/2614#issuecomment-2625112161

SKHRAPP, 2025-02-06:
@mrousavy Any update from this issue I using latest RN 0.77.0 getting this error

digitalbore, 2025-02-11:
Same problem here.

"react-native": "0.77.0",
"react-native-vision-camera": "^4.6.3"

kotlinVersion = "2.0.21"