margelo/react-native-vision-camera #1548: How to reduce the size of recorded videos on android ❓

alfaGefersona, 2023-04-03:
### Question

How to reduce the size of videos recorded on android ?, Currently a video of about 8 seconds has approximately 20Mb, even using different presets the size remains exaggeratedly large

### What I tried

I tried setting preset='vga-640x480',

I also tried to define videoCodec inside the RecordVideoOptions property, but even so the size is exaggeratedly large

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

garmoshka-mo, 2023-04-07:
Photo & video sizes not configurable neither with `presets` nor with `format` props. (With `format` you can actually change size - but it will be random, not even close to width/height of specified format).

I had to hardcode desired resolution with patch:
[react-native-vision-camera+2.15.4.patch](https://github.com/mrousavy/react-native-vision-camera/files/11178551/react-native-vision-camera%2B2.15.4.patch)

ratz6, 2023-04-12:
Should we unlink the video using RNFS after it's being uploaded to cloud ? All these recordings are taking up a lot of the app space

alfaGefersona, 2023-04-14:
> Photo & video sizes not configurable neither with `presets` nor with `format` props. (With `format` you can actually change size - but it will be random on android, not even close to width/height of specified format).
> 
> I had to hardcode desired resolution with patch: [#868 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/868#issuecomment-1502782934)

alfaGefersona, 2023-04-14:
Thanks @garmoshka-mo  , I will work around your solution and see what is the best approach to my problem .

 As you pointed out  This task https://github.com/mrousavy/react-native-vision-camera/issues/868. **Wrong format on ANDROID.**  it can be useful to solve this problem in trying to decrease the size of video files when using this camera library.
 
 Thanks a lot for the help!

closed 2023-04-14 by alfaGefersona

t4dek, 2023-09-27:
@mrousavy thx for releasing v3, it is awesome and solves a lot of issues.
But unfortunately  this is still an issue, videos are really large in terms of fileSize.

Previously it could have been fixed with patch https://github.com/mrousavy/react-native-vision-camera/issues/868#issuecomment-1502782934 
but because of rewrite of the Android this patch is not working anymore...

previously inside src/main/java/com/mrousavy/camera/CameraView.kt we could do next:

` videoRecorderBuilder.setQualitySelector(QualitySelector.from(Quality.SD))`

and it was working.

Can you please advise how to achieve the same behaviour with new implmentation? 
I am ready to hardcode SD quality in android codebase and make myself a patch, because I don't see other solution.

Thanks in advance!

t4dek, 2023-09-28:
@mrousavy I found a place where I can set bitrade of video, and decreasing the bitrade, decreased the size of output file. Would be good if we can pass it from React Native side :)

mrousavy, 2023-09-29:
Check out https://github.com/mrousavy/react-native-vision-camera/pull/1882 ;)

ussef11, 2024-02-14:
const devices = useCameraDevices();
  const _device = devices[1];

  const [device, setDevice] = useState(null);
  const [format, setFormat] = useState(null);

  useEffect(() => {
    if (_device) {
      setDevice(_device);
      console.log('formatformat', _device.formats);
      const desiredResolution = {width: 176, height: 144};

      const matchingFormat = _device.formats.find(
        f =>
          f.videoWidth === desiredResolution.width &&
          f.videoHeight === desiredResolution.height,
      );

      console.log('matching', matchingFormat);
      if (matchingFormat) {
        setFormat(matchingFormat);
      } else {
        console.warn('Desired resolution not supported');
      }
    }
  }, [_device]);          this solution  work for me

mrousavy, 2024-08-05:
As all other people you have spammed about this question already told you, 151 MB for a directly recorded video isn't too big for 1 minute.
This is normal.

You need to work with compression, e.g. use ffmpeg to get the file size down.

If you spam your question across 10 different unrelated issues again I'm going to have to ban you from the repo.

brianw-tech, 2025-04-15:
> [@mrousavy](https://github.com/mrousavy) I am using "react-native-vision-camera": "^4.4.2", "react-native": "^0.73.8",
> 
> const format = useCameraFormat(backCamera, [ { videoResolution: { width: 1280, height: 720 } }, { fps: 30 },{pixelFormat: 'native'} ])
> 
> await cameraRef.current?.startRecording({ videoCodec: 'h264', videoBitRate: 'extra-low', onRecordingFinished: video => { const path = video?.path setVideoPath(path) clearInterval(timerId); setCustomUrl(path) }, onRecordingError: error => { setIsRecording(false); console.error('Recording error: ', error); clearInterval(timerId); } }); While recording a video for 1 minute, the file size becomes as large as 151 MB. How can I reduce the file size? Please help.

Did you get a solution for this?