margelo/react-native-vision-camera #1550: 🐛 Black screen on Windows subsystem for Android.

mazmoiz, 2023-04-04:
### What were you trying to do?

I get a black screen on the camera preview when I run my react-native (Android) app on the Windows subsytem for Android.

### Reproduceable Code

```tsx
camera.current.startRecording({
      onRecordingFinished: (video) => {
          handleRecordingFinished(video);
      },
      onRecordingError: (error) => console.error(error),
});
```

### What happened instead?

There are two issues here:
1. Camera preview shows black screen.
2. App crashes when trying to start recording.

### Relevant log output

```shell
04-03 06:13:52.862   102   197 E GuestCameraDeviceSession: copyBuffer:1039 sizes mismatch [src ht:720 wt:1280] [dst ht:1080 wt:1920] 

04-03 06:13:52.924   102   197 E GuestCameraDeviceSession: Failed at device/microsoft/latte/hal/camera/GuestCameraDeviceSession.cpp:1042 (copyBuffer)
```

### Device

Windows subsystem for Android

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

mazmoiz, 2023-04-26:
This issue is finally resolved by setting the camera format:

`const format = useCameraFormat(device);`

```
<Camera
    ref={camera}
    device={device}
    video={true}
    format={format}
 />
```