margelo/react-native-vision-camera #1551: 🐛  Third-party Android NDK suffering from vulnerabilities to multiple CVE exploits

mekwall, 2023-04-06:
### What were you trying to do?

We recently concluded a penetration test of our app and found several vulnerabilities that we believe stems from the the third-party Android NDK used by `react-native-vision-camera`. The vulnerabilities exist in the following libraries: opencv (v3.4.5), libjpg-turbo (v1.5.3) and libpng (v1.6.35). We suggest that you investigate this and upgrade the third-party NDK and/or its dependencies to later versions that have been patched against these exploits.

Here's a list of the exploits we found:

## opencv

Dependency opencv version 3.4.5 was detected at lib/arm64-v8a/libNativeBridge.so

**CVE-2019-14492**: An issue was discovered in OpenCV before 3.4.7 and 4.x before 4.1.1. There is an out of bounds read/write in the function HaarEvaluator::OptFeature::calc in modules/objdetect/src/cascadedetect.hpp, which leads to denial of service. 

**CVE-2019-14491**: An issue was discovered in OpenCV before 3.4.7 and 4.x before 4.1.1. There is an out of bounds read in the function cv::predictOrdered<cv::HaarEvaluator> in modules/objdetect/src/cascadedetect.hpp, which leads to denial of service. 

**CVE-2019-19624**: An out-of-bounds read was discovered in OpenCV before 4.1.1. Specifically, variable coarsest_scale is assumed to be greater than or equal to finest_scale within the calc()/ocl_calc() functions in dis_flow.cpp. However, this is not true when dealing with small images, leading to an out-of-bounds read of the heap-allocated arrays Ux and Uy.

## libjpeg-turbo

Dependency libjpeg-turbo version 1.5.3 was detected at lib/arm64-v8a/libnative-imagetranscoder.so and suffers from the following vulnerabilities:

**CVE-2018-14498**: get_8bit_row in rdbmp.c in libjpeg-turbo through 1.5.90 and MozJPEG through 3.3.1 allows attackers to cause a denial of service (heap-based buffer over-read and application crash) via a crafted 8-bit BMP in which one or more of the color indices is out of range for the number of palette entries. CVE-2018-20330: The tjLoadImage function in libjpeg-turbo 2.0.1 has an integer overflow with a resultant heap-based buffer overflow via a BMP image because multiplication of pitch and height is mishandled, as demonstrated by tjbench.

## libpng

Dependency libpng version 1.6.35 was detected at lib/arm64-v8a/libNativeBridge.so and suffers from the following vulnerabilities:

**CVE-2019-7317**: png_image_free in png.c in libpng 1.6.x before 1.6.37 has a use-after-free because png_image_free_function is called under png_safe_execute.

### What happened instead?

Not applicable.

### Device

Android

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

mrousavy, 2023-08-18:
Hey - thanks for reporting. What's the solution? Updating the ndkVersion?

closed 2023-09-01 by mrousavy