margelo/react-native-vision-camera #1554: 🐛 "Camera View" is not Fabric compatible yet

gokul1099, 2023-04-10:
### What were you trying to do?

I am using vision camera in rn 0.71.4. There I am, getting this error message. Is that possible to run the vision camera with fabric enabled? 

[image]

### Reproduceable Code

```tsx
I used the code from the demo application in this repo.
```

### What happened instead?

Camera not opening because of this.

### Device

Pixel 5(Android 11), Nokia 5.3(Android 12)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EazycTee, 2023-06-09:
same issue... need help

robvigilleadingedje, 2023-06-23:
Anyone find a solution to this?  App requires new RN architecture..  Anyone know how to disable fabric but keep new architecture?

luvnishCapgemini, 2023-06-26:
I am facing this same issue. Any workaround?

ravindraguptacapgemini, 2023-06-27:
@mrousavy please help suggest if the camera feature is not available yet for android? Getting same error "Camera View" is not fabric compatible yet. Also if there is any workaround we can try to still run camera with fabric on both android and iOS. Thanks

ace0109, 2023-07-08:
[image]
The same problem in the IOS is no problem

ace0109, 2023-07-08:
> [image] The same problem in the IOS is no problem

newArchEnabled=false
I set this configuration to false first, because it is not compatible, when will it be compatible?

mrousavy, 2023-07-10:
Just like all 5 of you discovered, VisionCamera is not yet fabric compatible. 

I didn't work on making it Fabric compatible yet. If you want me to make it run on Fabric, either wait until I have the free time to do so, or pay me to do it: [sponsor me on GitHub](https://github.com/sponsors/mrousavy) :)

DominikHinc, 2023-08-09:
Hey,
I have prepared an initial support for fabric: https://github.com/mrousavy/react-native-vision-camera/pull/1684

closed 2023-09-01 by mrousavy

keremkurtulus, 2023-09-02:
I overcame the issue with the help of this discussion article. 
https://github.com/reactwg/react-native-new-architecture/discussions/135

It may not be an absolute fix but it doesn't hurt to try.

I created `react-native.config.js` file and updated it like here; if you want to know why I use this config file find the discussion article above.
```
module.exports = {
  project: {
    android: {
      unstable_reactLegacyComponentNames: ['CameraView'],
    },
    ios: {
      unstable_reactLegacyComponentNames: ['CameraView'],
    },
  },
};

```

ravindraguptacapgemini, 2023-11-16:
> module.exports = {
>   project: {
>     android: {
>       unstable_reactLegacyComponentNames: ['CameraView'],
>     },
>     ios: {
>       unstable_reactLegacyComponentNames: ['CameraView'],
>     },
>   },
> };

@mrousavy  Somehow I am able to build the library with this inter-op layer change with Fabric on React Native 0.72.6, but still I see few issues with Events, please

turkergercik, 2023-12-09:
> > module.exports = {
> > project: {
> > android: {
> > unstable_reactLegacyComponentNames: ['CameraView'],
> > },
> > ios: {
> > unstable_reactLegacyComponentNames: ['CameraView'],
> > },
> > },
> > };
> 
> @mrousavy Somehow I am able to build the library with this inter-op layer change with Fabric on React Native 0.72.6, but still I see few issues with Events, please check the screenshot ![MicrosoftTeams-image](https://private-user-images.githubusercontent.com/97147467/283543233-4a1499ba-587b-43f4-93bd-0c05873f00f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMjg5OTcsIm5iZiI6MTcwMjEyODY5NywicGF0aCI6Ii85NzE0NzQ2Ny8yODM1NDMyMzMtNGExNDk5YmEtNTg3Yi00M2Y0LTkzYmQtMGMwNTg3M2YwMGY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA5VDEzMzEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyMzgwMzhlY2NmNDQ5ZmQ1ZjEyZWQyMjI1MmEzNTQ3YTE5Y2RmNjM2ZDA3NGUwNzZiZmNjMWRmYWEyODk3MmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ossgkQz3DK8qPV5gb5lkBLUD2heVmREiltC1AKRfkOU)

same for me

fahadhaque007, 2023-12-24:
Any one manage to fix this issue?

BaseMax, 2024-01-19:
Same issue. Any updates?

[image]

mrousavy, 2024-01-22:
https://github.com/mrousavy/react-native-vision-camera/issues/1554#issuecomment-1628498006

nonam4, 2024-01-26:
I'm using vision camera `3.8.2` and still recieving this error message, isn't this new version supposed to have fabric support?

nonam4, 2024-01-26:
also waiting for a fix

farwayer, 2024-02-15:
@ravindraguptacapgemini @turkergercik
https://github.com/mrousavy/react-native-vision-camera/pull/2569

CarterSnich, 2025-02-06:
What is Fabric? I'm getting this issue, but the camera worked before.