margelo/react-native-vision-camera #1558: 🐛 StopRecording doesn't emit any event on specific android devices (finish or error)

angelos3lex, 2023-04-12:
### What were you trying to do?

I try to record video. Some specific android devices do not fire any callback (neither `onRecordingError` nor `onRecordingFinished`) after calling `camera.stopRecording()`. I don;t even get any callback through the `Camera` component in the `onError` prop.
This is happening no matter the size of the video (even on 2-3 sec video)
The only device I have reproduced at the moment (100% reproduce, happening all the time, using either front or back camera) is:
```
Brand: Samsung
Model: Galaxy Tab A8
Version: Android 13
RAM free: 778.43 MB
Disk free: 39.77 GB 
```

### Reproduceable Code

```tsx
The way i choose camera is:

const devices = useCameraDevices();
let device;
if (props.useFrontCamera) {
    device = devices.front;
} else {
    device = devices.back;
}
```
```

### What happened instead?

I get no callback. The app continues to operate as normal, no crash or something, just no video is recorded and i don't get any event.

### Device

Samsung Galaxy Tab A8 (android)

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rosendosalazarbta, 2023-04-17:
I have the same issue.

pranav-lio, 2023-04-18:
@angelos3lex @rosendosalazarbta 
Can you check if you are taking microphone and camera permission before starting recording. 
I faced this same issue 

Just added permission for microphone and camera and it starting working.

angelos3lex, 2023-04-19:
@pranav-lio not the case for me, adding&asking microphone permisions don't fix it. also I don't use audio.

DnEgorWeb, 2023-05-16:
Happens for me as well on Android 9. Android 10 works fine, haven't checked others yet

DnEgorWeb, 2023-05-16:
@angelos3lex have you found a workaround?

angelos3lex, 2023-05-16:
Unfortunately no, still an issue

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

angelos3lex, 2023-10-06:
I can verify this is fixed in V3.3.1 that we checked! Nice job @mrousavy! 

_We have some other crash on some cases but ill open a new issue once we find out more about how to reproduce this._
_Till then, i just leave here the trace in case you have any idea._
```
Fatal Exception: java.lang.IllegalArgumentException: Surface was abandoned
       at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:135)
       at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:615)
       at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:325)
       at com.mrousavy.camera.core.outputs.SurfaceOutput.toOutputConfiguration(SurfaceOutput.kt:38)
       at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:57)
       at com.mrousavy.camera.core.CameraSession.getCaptureSession(CameraSession.kt:438)
       at com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:532)
       at com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49)
       at com.mrousavy.camera.core.CameraSession$startRunning$1.invokeSuspend(:14)
```

mrousavy, 2023-10-06:
Awesome, thanks!

Yea this error might be the Preview that does something wrong, I'll need a bug report, logs & repro :)