margelo/react-native-vision-camera #1563: Upgrade v2 to reanimated v3❓

angelica-snowit, 2023-04-19:
### Question

Reanimated v2 has a lot of problems (example: https://github.com/software-mansion/react-native-reanimated/issues/4341).
While we wait for react native vision camera v3, can you please upgrade the v2 to reanimated v3?

### VisionCamera Version

v2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

fukemy, 2023-05-24:
v3 still have a lot of problems

EwaGuziejko, 2023-06-23:
But it's better than v2, it would be great if we could get an upgrade ;)

felipemillhouse, 2023-06-28:
So does it mean that I cannot go to Reanimated 3 in my project?

I'm asking that because we are upgrading to RN 0.72.0 and we got a few errors to build because of Reanimated lib, and then we upgraded it to v3 and the error went away, but now we are having problems with `react-native-vision-camera` on file `Frame Processor/FrameProcessorRuntimeManager` at line 146
`auto worklet = reanimated::ShareableValue::adapt(runtime, arguments[1], runtimeManager.get());`
error: "No member named 'ShareableValue' in namespace 'reanimated'"

Can we use `react-native-vision-camera` just with Reanimated v2?

liveforownhappiness, 2023-07-06:
> 

any updated?

jimitshah-dp, 2023-07-07:
Thanks for the awesome work in react-native-vision-camera, are we planning to incorporate this?

RP-alissonpaschoal, 2023-08-07:
@felipemillhouse currently we cant use it with reanimated v3. With VisionCamera v3 it will be possible, and it wont be required to have reanimated installed anymore. But the development of v3 is still in progress https://github.com/mrousavy/react-native-vision-camera/issues/1376

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (or V2 here) (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy