margelo/react-native-vision-camera #1564: Is it possible to display the front and back cameras at the same time❓

lsmith-scorchsoft, 2023-04-21:
### Question

We are trying to display a `<Camera /> `view for both the front and back physical devices simultaneously, but only the last one to load is displayed and updated. Is there some config that we're missing or is it not possible with this package? If it's not possible is this a native limitation or just due to the package's implementation?

Example setup;

```
import React from 'react';
import { View } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

const SimulCameras = () => {
    const devices = useCameraDevices();
    const { front, back } = devices;

    if (front == null || back == null) {
        return null;
    }

    return (
        <View style={{ width: '100%', height: '100%' }}>
            <Camera
                style={{ width: '100%', height: '50%' }}
                device={front}
                isActive />
            <Camera
                style={{ width: '100%', height: '50%' }}
                device={back}
                isActive />
        </View>
    );
};

export default SimulCameras;
```

### What I tried

- Using multiple hooks
- separating to different components

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

t4dek, 2023-07-11:
@lsmith-scorchsoft do you have any updates on this issue? I am facing a similar one...

dehimer, 2023-08-28:
Actual. I also desired to use two cameras at same time, but only one is working

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; this might be possible on Android, but not on iOS because on iOS you need a completely separate MultiCamera session

closed 2023-09-30 by mrousavy

petrohusak, 2024-01-15:
Hello @mrousavy. It seems that this still isn't possible to do this on iOS devices (version 3.7.1). Do you know any workaround how it can be done?