margelo/react-native-vision-camera #1565: Runtime JAR files in the classpath should have the same version. 🐛

chris2kus31, 2023-04-22:
### What were you trying to do?

After installing the library and then trying to `npx react-native run-android  --verbose` the build fails.  

### Reproduceable Code

```tsx
I am trying to install 2.15.4 v or react-native-vision-camera

What I am running

"react-native": "0.64.2",

buildscript {
    ext {
        buildToolsVersion = "29.0.3"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 31
        ndkVersion = "20.1.5948944"
    }
    repositories {
        google()
       mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.1.0")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
```

### What happened instead?

What I end up getting is this error msg below.   and the build fails.   

```
> Task :react-native-vision-camera:compileDebugKotlin FAILED
w: Runtime JAR files in the classpath should have the same version. These files were found in the classpath:
    /Users/chuy/.gradle/caches/transforms-2/files-2.1/a7540161cf872bf6844fda8893063aa6/jetified-kotlin-stdlib-jdk8-1.5.30.jar (version 1.5)
    /Users/chuy/.gradle/caches/transforms-2/files-2.1/647b9fbf8b05a891d909b37be844f76e/jetified-kotlin-stdlib-jdk7-1.5.30.jar (version 1.5)
    /Users/chuy/.gradle/caches/transforms-2/files-2.1/44d9f8bcd96348437000c79944fd3d45/jetified-kotlin-stdlib-1.6.10.jar (version 1.6)
    /Users/chuy/.gradle/caches/transforms-2/files-2.1/7b1b54c1d73181683b14469ad8453a4c/jetified-kotlin-stdlib-common-1.6.10.jar (version 1.6)
w: Some runtime JAR files in the classpath have an incompatible version. Consider removing them from the classpath
e: /Users/chuy/Desktop/coc-mobile-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (87, 116): Unresolved reference: resolveView
e: /Users/chuy/Desktop/coc-mobile-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt: (65, 76): Unresolved reference: resolveView
```

### Device

Android build

### VisionCamera Version

2.15.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EmanoelLoreto, 2023-05-25:
Has anyone managed to fix it?

poltak, 2023-08-17:
#957 solved it for me. i.e., downgrading to 2.13.0 as later version have a dep on a later version of RN than I am using

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy