margelo/react-native-vision-camera #1567: 🐛 build error on iOS

vbylen, 2023-04-23:
### What were you trying to do?

attempting to build my project

### What happened instead?

I'm receiving the following error when trying to build my project:

### Relevant log output

```shell
node_modules/react-native-vision-camera/ios/Skia Render Layer/SkiaMetalCanvasProvider.h:13:10)

  11 | 
  12 | #include <functional>
> 13 | #include <include/gpu/GrDirectContext.h>
     |          ^ 'include/gpu/GrDirectContext.h' file not found
  14 | #include <mutex>
  15 | #include <memory>
  16 | #include <atomic>
```

### Device

ios

### VisionCamera Version

3.0.0-rc.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

R4tch3t, 2023-05-01:
same error how fix these headers no founds?

vbylen, 2023-05-01:
@R4tch3t see https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1525731923 for a (temporary) potential solution

R4tch3t, 2023-05-01:
> @R4tch3t see [#1376 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1525731923) for a (temporary) potential solution

Thanks work for me!

AnomalousLLC, 2023-05-08:
@vbylen Would you be able to share a template for Expo? Trying to recreate the same and running into issues

vbylen, 2023-05-08:
@AnomalousLLC you just need to install `patch-package` and create a file under your patches folder in which you then copy/paste the code shared above.

[image]

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy