margelo/react-native-vision-camera #1568: Not building for RN 0.64.2🐛

chris2kus31, 2023-04-23:
### What were you trying to do?

Build android 

### Reproduceable Code

```tsx
When running `npx react-native run-android  --verbose` build fails 

RN 0.64.2
React animated 2.12.0
```

### What happened instead?

* Where:
Build file '/Users/chuy/Desktop/coc-mobile-app/node_modules/react-native-vision-camera/android/build.gradle' line: 473

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
> Cannot expand ZIP '/Users/user/Desktop/app/node_modules/react-native-reanimated/android/react-native-reanimated-64-jsc.aar' as it does not exist.

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

### Device

android build

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

RuFang-21, 2023-04-25:
Same issues here , any solution for this ? 

thanks !

sparkison, 2023-06-28:
For anyone still landing here, I was able to manually include RN Reanimated lib that contains the required file to fix the problem. Version 1.8.0 and 1.7.0 contain the required file, version 1.9.0+ does not.

TLDR; `npm i react-native-reanimated@2.8.0 && npm i react-native-vision-camera@2.13.0` should do the trick 👍 

Currently works with RN 0.64.4 – this combination also works with iOS

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy