margelo/react-native-vision-camera #1574: Best way to draw that accounts for aspect ratio❓

eledahl, 2023-04-26:
### Question

Hey hey,

I am trying to draw a pose with this repo. It works great for normal 16:9 iphones but when someone uses an phone pro max which is 19.5:9 then the pose doesn't grow with the screen. 

[image]

This is my current function for rendering the pose: 

```
const renderPose = () => {
    if (poses != null && poses.length > 0) {
      const selectedJoints = [
        poses[0].LeftShoulder,
        poses[0].RightShoulder,
        poses[0].LeftHip,
        poses[0].RightHip,
        poses[0].RightKnee,
        poses[0].LeftKnee,
        poses[0].RightAnkle,
        poses[0].LeftAnkle,
        poses[0].RightShoulder,
        poses[0].LeftToe,
        poses[0].RightToe,
        poses[0].RightElbow,
        poses[0].LeftElbow,
        poses[0].RightThumb,
        poses[0].LeftThumb,
      ]
        .filter((k) => (k.inFrameLikelihood ?? 0) > 0.3)
        .map((k, index) => {
          // Flip horizontally on android or when using back camera on iOS.
          return (
            <Svg >
              <Circle
                key={`skeletonkp_${k.type}`}
                cx={k.position.x}
                cy={k.position.y}
                r={25}
                strokeWidth="2"
                fill="transparent"
                stroke={Theme.COLORS.WHITE}
              />

              {/* <Circle
                key={`skeletonkp_${k.type}${index}`}
                cx={k.position.x}
                cy={k.position.y}
                r={10}
                strokeWidth="0"
                fill="#FAF9F6"
                stroke="transparent"
              /> */}
            </Svg>
          );
        });
      return (
        <Svg
          // height={Dimensions.get("window").height}
          // width={Dimensions.get("window").width}
          style={styles.svg}
          viewBox={getViewBox()}
        >
          {selectedJoints}
        </Svg>
      );
    } else {
      return <View></View>;
    }
  };

...
...
  svg: {
    position: "absolute",
    top: 0,
    left: 0,
    height: Dimensions.get("screen").height,
    width: Dimensions.get("screen").width,
    zIndex: 1,
}
```

### What I tried

I've tried adding in the output widths and preview widths as I had before with [tfjs](https://github.com/tensorflow/tfjs-examples/blob/master/react-native/pose-detection/App.tsx#L161) but that didn't show any points at all. When I was first using tfjs, the pose worked well with all screen sizes.

### VisionCamera Version

"^2.15.2"

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

eledahl, 2023-04-30:
This _may_ actually be a bug. When I console log the frame height and width of the iphone pro max, the ratio is still 16:9. Looking through the formats, it doesn't appear there are any 19.5:9 options that even appear.

Michal41, 2023-08-14:
Hi, i have similar problem, did you resolve it ?

closed 2023-08-14 by eledahl