margelo/react-native-vision-camera #1585: 🐛 takePhoto() not always resolving Android

marcpropeller, 2023-05-04:
### What were you trying to do?

Trying to take a picture using takePhoto. There does seem to be similar bugs reported, but they seem to be talking about emulators and frame processing. I am just trying to take a basic photo. The component is conditionally rendered on a page.

### Reproduceable Code

```tsx
const camera = useRef<Camera>(null);
const [initialising, setInitialising] = useState<boolean>(true);
const [flash, setFlash] = useState<'off' | 'on'>('off');

const takePhoto = useCallback(() => {
	console.log('take photo');
	if (camera.current) {
		if (!initialising) {
			camera?.current
				?.takePhoto({
					flash,
					enableAutoStabilization: true,
				})
				.then(async (file: PhotoFile) => {
					console.log('taken photo');
					console.log(file);
					if (props.onPhotoTaken !== undefined) {
						props.onPhotoTaken(file);
					}
					setSaving(false);
				})
				.catch(e => {
					console.log(e);
				});
		} else {
			Alert.alert(
				'Camera Error',
				'Camera not initialised. Please try again.',
				[
					{
						text: 'Close',
						onPress: () => console.log('Cancel Pressed'),
						style: 'cancel',
					},
				],
				{ cancelable: false }
			);
		}
	} else {
		Alert.alert(
			'Camera Error',
			'Camera reference pending. Please try again.',
			[
				{
					text: 'Close',
					onPress: () => console.log('Cancel Pressed'),
					style: 'cancel',
				},
			],
			{ cancelable: false }
		);
	}
}, [initialising]);

const onError = useCallback((error: CameraRuntimeError) => {
	console.log('Camera error!');
	console.error(error);
}, []);

const onCameraInitialized = useCallback(() => {
	console.log('Camera initialized!');
	setInitialising(false);
}, []);

return (
	<>
		<Camera
			onError={onError}
			style={StyleSheet.absoluteFillObject}
			device={device}
			isActive={isFocused}
			onInitialized={onCameraInitialized}
			photo
			ref={camera}
		/>
		<View style={styles.captureButton}>
			<TouchableOpacity
				onPress={() => {takePhoto();}}
				disabled={initialising}>
				<View style={styles.shadow} />
				<View
					style={[
						styles.captureInnerButton,
						{
							backgroundColor: recording
								? 'red'
								: 'transparent',
						},
					]}
				/>
			</TouchableOpacity>
		</View>
	</>
);
```

### What happened instead?

Seems to ges stuck in some sort of loop with the SurfaceView GC. It doesn't get to the 'taken photo' console.log. On other android phones it seems to work, what's odd is that I've tried this on 3 Xiaomi devices, and it happens on 2 of them (works on Xiaomi Redmi Note 11 Pro (android 12 SKQ1.211006.001, MIUI 13.0.5)). Also tested on a variety of Samsung devices and works.
It will occasionally work, if I take enough photos. Can't seem to find what scenario makes it work as it seems different every time I try. If you need any more information, let me know.

[logcat.log](https://github.com/mrousavy/react-native-vision-camera/files/11397402/logcat.log)

### Relevant log output

```shell
See attachment above - going over character limit.
```

### Device

Xiaomi Mi 10T (Android 12 SKQ1.211006.001, MIUI 14.0.1), Xiaomi Redmi (Android 10 QKQ1.191215.002, MIUI 12.0.6)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mihailo-misic, 2023-05-26:
@mrousavy We are also facing an issue that seems very similar.

On some Android phones the `takePhoto` is called but the promise returned does not resolve.

The problem was reported on the following devices:
- Gigaset GX 290
- Huawei P40 Pro

Since this is a big concern for us, we are ready to sponsor the prioritization of this issue.
How much would it cost and by when could we expect the fix?

edritech93, 2023-07-05:
I found this issue on the emulator Android 10

sinau123, 2023-07-18:
same here, on Xiaomi Mi A1 android 9

brunezkey, 2023-07-25:
We're having a similar issue over here.

We're seeing reports of this issue on a daily basis from our application error monitoring system:

`[unknown/unknown: [unknown/unknown] Camera is closed.]`. This issue is occurring exclusively on Android devices.

We've attempted to reproduce the error using various emulators without success. The affected devices include, but are not limited to:

- `Samsung Galaxy S22 Ultra`
- `Samsung Galaxy S21`
- `Samsung Galaxy S3`
- `Samsung Exynos 990`
- `Moto G Stylus 5G`

We're currently investigating the issue and seeking more information to help resolve it. Any extra information you could provide to help us track this issue would be greatly appreciated.

mrousavy, 2023-07-31:
Hm, maybe V3 fixes this

marcpropeller, 2023-07-31:
> Hm, maybe V3 fixes this

Hi @mrousavy, 

Thanks for the update. Is the V3 branch in a position that it can be used and are there any breaking changes we need to be aware of? If/when it's good to go, I'll try upgrading and re-test the issue.

Thanks,

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

chinnuu05, 2024-07-15:
I'm still getting this issue using webcam0, it is not fixed

tomtargosz, 2024-08-05:
I'm seeing this as well.

OS: Android 14
Phone: SAMSUNG Galaxy A13
Library version: 4.2.0

mczwalker, 2024-08-13:
Same problem here.

Phone: SAMSUNG GALAXY A21s
Android: 11
Library: 4.5.1,
React Native: 0.74.4

SwhiteMHC, 2024-08-21:
I am also experiencing this issue.

Phone: Samsung Galaxy A14 5G
Android: 14
Library: 4.5.1
React Native: 0.74.5

samuel-andres, 2024-09-17:
Same here:

```
Phone: Samsung Galaxy A33 5G (SM-A336B)
Android: 14
Library: 4.5.0
React Native: 0.73.7

```
and also

```
Phone: Samsung Galaxy Tab S8 Ultra (SM-X700)
Android: 14
Library: 4.5.0
React Native: 0.73.7
```

phi0411141, 2025-04-13:
I had `Camera is closed` issue and fixed it by not setting `isActive={false}` (pause) when taking the photo

cccolo, 2025-04-22:
> I had `Camera is closed` issue and fixed it by not setting `isActive={false}` (pause) when taking the photo

do you have some example?

phi0411141, 2025-04-23:
> > I had `Camera is closed` issue and fixed it by not setting `isActive={false}` (pause) when taking the photo
> 
> do you have some example?

In my case, when user clicks taking photo, I set `<Camera isActive={false}/>` to pause the camera, but that would cause `Camera is closed` on android, especially samsung phone, the reason is that the Camera had not done taking the image, so just make sure `<Camera isActive={true} /> when taking the photo