margelo/react-native-vision-camera #1591: How to use the `1.8.0` Kotlin version❓

adscud, 2023-05-06:
### Question

How to use the `1.8.0` Kotlin version ? 

Today, when we try to use this version we have this following error : 

```
> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: The Android extensions ('kotlin-android-extensions') compiler plugin is no longer supported. Please use kotlin parcelize and view binding.
```

Most of librairies with RN > 71 support uses `1.8.0` version of Kotlin so I can't use the `1.6.20` for example :/

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

closed 2023-05-06 by adscud

christophemenager, 2023-05-16:
I have the same issue! @adscud how did you manage to fix this issue?

adscud, 2023-05-17:
@christophemenager 
Hello, 

In the `android/build.gradle`

```
buildFeatures {
    viewBinding = true
    if (REACT_NATIVE_VERSION >= 71) {
      prefab true
    }
  }
```

kyle-belle, 2023-06-04:
what if im using expo?

RRaideRR, 2023-06-30:
> what if im using expo?

You can use patch-package to patch the build.gradle file in node_modules/react-native-vision-camera/android/src folder as @adscud described.

pauldcollins, 2023-07-03:
I'm also experiencing this problem. Have tried the above steps, but I feel I'm going round in circles now! Here's the initial error I get:

`> Task :react-native-vision-camera:compileDebugKotlin FAILED` 

I had already deleted the "apply plugin" lines in `build.gradle`, suggested above and tried to add the `buildFeatures` object, but then I get the following error:

`Could not find method buildFeatures() for arguments [build_5euaw2kj96akw3m9uzpd5dunz$_run_closure1$_closure5@19b82ba4] on object of type org.gradle.api.internal.initialization.DefaultScriptHandler.`

And if I delete that, I get this error:

`Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.`
`> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction`

I'm using `Kotlin 1.8.0` (in **gradle.properties.**). I did try rolling back to `1.6.20`, but then I get this error: 

`The Kotlin Gradle plugin was loaded multiple times in different subprojects, which is not supported and may break the build.
This might happen in subprojects that apply the Kotlin plugins with the Gradle 'plugins { ... }' DSL if they specify explicit versions, even if the versions are equal.
Please add the Kotlin plugin to the common parent project or the root project, then remove the versions in the subprojects.
If the parent project does not need the plugin, add 'apply false' to the plugin line.
The Kotlin plugin was loaded in the following projects: ':react-native-gesture-handler', ':react-native-safe-area-context', ':react-native-screens', ':react-native-vision-camera'`

Henriquehlr, 2023-08-11:
@pauldcollins I have the same problem as you, did you manage to solve it?

pauldcollins, 2023-08-11:
Hi @Henriquehlr sadly no I have not solved it. I parked the Android build for now. Would love to find a solution!