margelo/react-native-vision-camera #1592: 🐛 Build error in React Native IOS

pauloscardoso, 2023-05-14:
### What were you trying to do?

Install lib. I created a reusable camera component and it's working perfectly on Android. But not build on ios.

Xcode version: 14.3|
React Native version: 0.71.8

### What happened instead?

The software is not built correctly in the ios simulator

### Relevant log output

```shell
info Opening the app on iOS...
info Found Xcode workspace "appointment.xcworkspace"
info Found booted iPhone SE (3rd generation)
info Launching iPhone SE (3rd generation)
info Building (using "xcodebuild -workspace appointment.xcworkspace -configuration Debug -scheme appointment -destination id=4775FBD3-C8EA-49E9-90C4-72BCE9B64D2C")
Command line invocation:
    /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -workspace appointment.xcworkspace -configuration Debug -scheme appointment -destination id=4775FBD3-C8EA-49E9-90C4-72BCE9B64D2C

User defaults from command line:
    IDEPackageSupportUseBuiltinSCM = YES

Prepare packages

Computing target dependency graph and provisioning inputs

Create build description

note: Building targets in dependency order
ClangStatCache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang-stat-cache /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator16.4.sdk /Users/paulocardoso/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphonesimulator16.4-20E238-.sdkstatcache
    cd /Users/paulocardoso/code/appointment-stable/ios
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang-stat-cache /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator16.4.sdk -o /Users/paulocardoso/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphonesimulator16.4-20E238-.sdkstatcache

SwiftDriver\ Compilation\ Requirements vision-camera-code-scanner normal arm64 com.apple.xcode.tools.swift.compiler (in target 'vision-camera-code-scanner' from project 'Pods')
    cd /Users/paulocardoso/code/appointment-stable/ios/Pods
    builtin-Swift-Compilation-Requirements -- /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc -module-name vision_camera_code_scanner -Onone -enforce-exclusivity\=checked @/Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/Objects-normal/arm64/vision-camera-code-scanner.SwiftFileList -DDEBUG -D COCOAPODS -Xcc -fmodule-map-file\=/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/FBLPromises/PromisesObjC.modulemap -Xcc -fmodule-map-file\=/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React/React-Core.modulemap -Xcc -fmodule-map-file\=/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/folly/RCT-Folly.modulemap -Xcc -fmodule-map-file\=/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/yoga/Yoga.modulemap -import-underlying-module -Xcc -fmodule-map-file\=/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/vision_camera_code_scanner/vision-camera-code-scanner.modulemap -sdk /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator16.4.sdk -target arm64-apple-ios11.0-simulator -enable-bare-slash-regex -g -module-cache-path /Users/paulocardoso/Library/Developer/Xcode/DerivedData/ModuleCache.noindex -Xfrontend -serialize-debugging-options -enable-testing -index-store-path /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Index.noindex/DataStore -swift-version 5 -I /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Products/Debug-iphonesimulator/vision-camera-code-scanner -I /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Products/Debug-iphonesimulator/GoogleMLKit -F /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Products/Debug-iphonesimulator/vision-camera-code-scanner -F /Users/paulocardoso/code/appointment-stable/ios/Pods/MLImage/Frameworks -F /Users/paulocardoso/code/appointment-stable/ios/Pods/MLKitBarcodeScanning/Frameworks -F /Users/paulocardoso/code/appointment-stable/ios/Pods/MLKitCommon/Frameworks -F /Users/paulocardoso/code/appointment-stable/ios/Pods/MLKitVision/Frameworks -F /Users/paulocardoso/code/appointment-stable/ios/Pods/hermes-engine/destroot/Library/Frameworks/universal -F /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Products/Debug-iphonesimulator/XCFrameworkIntermediates/hermes-engine/Pre-built -c -j8 -enable-batch-mode -incremental -Xcc -ivfsstatcache -Xcc /Users/paulocardoso/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphonesimulator16.4-20E238-.sdkstatcache -output-file-map /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/Objects-normal/arm64/vision-camera-code-scanner-OutputFileMap.json -use-frontend-parseable-output -save-temps -no-color-diagnostics -serialize-diagnostics -emit-dependencies -emit-module -emit-module-path /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/Objects-normal/arm64/vision_camera_code_scanner.swiftmodule -Xcc -I/Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/swift-overrides.hmap -Xcc -iquote -Xcc /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/vision-camera-code-scanner-generated-files.hmap -Xcc -I/Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/vision-camera-code-scanner-own-target-headers.hmap -Xcc -I/Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/vision-camera-code-scanner-all-non-framework-target-headers.hmap -Xcc -ivfsoverlay -Xcc /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/all-product-headers.yaml -Xcc -iquote -Xcc /Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/vision-camera-code-scanner.build/vision-camera-code-scanner-project-headers.hmap -Xcc -I/Users/paulocardoso/Library/Developer/Xcode/DerivedData/appointment-dbqkjxiqumcqckezuzixdckydxso/Build/Products/Debug-iphonesimulator/vision-camera-code-scanner/include -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Private -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Private/vision-camera-code-scanner -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/DoubleConversion -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/PromisesObjC -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/RCT-Folly -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-Core -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-callinvoker -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-cxxreact -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-hermes -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-jsi -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-jsiexecutor -Xcc -I/Users/paulocardoso/code/appointment-stable/ios/Pods/Headers/Public/React-jsinspector -Xcc -I/Users/pauloc [... cut ...]
Iphone SE 3rd Generation iOS Version 16.4

### VisionCamera Version

V4.2.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Pacmard, 2023-05-17:
Bumping it as I'm also running into issue with XCode 14.3 incompatibility

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy