margelo/react-native-vision-camera #1593: 🐛 FrameProcessor.mm - RNSkiOSPlatformContext - @Shopify/react-native-skia v0.1.188+

AnomalousLLC, 2023-05-14:
### What were you trying to do?

https://github.com/mrousavy/react-native-vision-camera/blob/v3/ios/Frame%20Processor/FrameProcessorUtils.mm#L33

[image]

RNSkiOSPlatformContext changed in 0.1.188: https://github.com/Shopify/react-native-skia/pull/1549

### What happened instead?

Build error

### Relevant log output

```shell
› Compiling react-native-vision-camera Pods/VisionCamera » FrameProcessorUtils.mm

❌  (/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS16.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:298:37)

  296 |         allocator_traits<_TpAlloc>::construct(__tmp, __get_elem(), _VSTD::forward<_Args>(__args)...);
  297 | #else
> 298 |         ::new ((void*)__get_elem()) _Tp(_VSTD::forward<_Args>(__args)...);
      |                                     ^ no matching constructor for initialization of 'RNSkia::RNSkiOSPlatformContext'
  299 | #endif
  300 |     }
  301 |
```

### Device

iOS

### VisionCamera Version

v3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pvedula7, 2023-05-24:
@AnomalousLLC 
This error got fixed with this PR: https://github.com/Shopify/react-native-skia/pull/1549.
New version will fix this issue, but a new issue arises regarding a missing GrRecordingContext.h as discussed [here](https://github.com/mrousavy/react-native-vision-camera/pull/1466#issuecomment-1553381626).

AnomalousLLC, 2023-05-24:
@pvedula7 That PR is referenced in the original post, it's the reason for this error. 

The issue regarding GrRecordingContext.h has already been discussed & resolved with a patch.

pvedula7, 2023-05-24:
@AnomalousLLC 
Do you have a link to the patch? I'm running the latest release on worklets and getting the GrRecordingContext error.

AnomalousLLC, 2023-05-24:
> @AnomalousLLC Do you have a link to the patch? I'm running the latest release on worklets and getting the GrRecordingContext error.

`--- a/react-native-skia.podspec
+++ b/react-native-skia.podspec
@@ -44,6 +44,31 @@ Pod::Spec.new do |s|
     "cpp/**/*.{h,cpp}"
   ]

+  s.subspec 'SkiaHeaders' do |ss|
+    ss.header_mappings_dir = 'cpp/skia'
+    ss.source_files = "cpp/skia/**/*.{h,cpp}"
+  end
+
+  s.subspec 'Utils' do |ss|
+    ss.header_mappings_dir = 'cpp/utils'
+    ss.source_files = "cpp/utils/**/*.{h,cpp}"
+  end
+
+  s.subspec 'Jsi' do |ss|
+    ss.header_mappings_dir = 'cpp/jsi'
+    ss.source_files = "cpp/jsi/**/*.{h,cpp}"
+  end
+
+  s.subspec 'Api' do |ss|
+    ss.header_mappings_dir = 'cpp/api'
+    ss.source_files = "cpp/api/**/*.{h,cpp}"
+  end
+
+  s.subspec 'RNSkia' do |ss|
+    ss.header_mappings_dir = 'cpp/rnskia'
+    ss.source_files = "cpp/rnskia/**/*.{h,cpp}"
+  end
+
   s.dependency "React"
   s.dependency "React-callinvoker"
   s.dependency "React-Core"`

AnomalousLLC, 2023-05-24:
https://github.com/mrousavy/react-native-vision-camera/issues/1376

AnomalousLLC, 2023-05-24:
> @AnomalousLLC Do you have a link to the patch? I'm running the latest release on worklets and getting the GrRecordingContext error.

https://github.com/mrousavy/react-native-vision-camera/issues/1376

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy